sharpie7/circuitjs1 #307: Feture: Add manual V/div scope settings

E3V3A, 2019-02-27:
It's a bit of a nuisance to see a stacked scope trace where one output ranges from 0-12V and the other between 0-5V, but the scope show the 5V trace as being twice as high. 

Please consider adding a manual override with a slider for the scope `V/div` setting like this:

[image]

kesterlester, 2020-05-10:
+1   from me.

Of all the feature requests sitting on this site, my guess is this is the one that will help most 
people most of the time.

While automatic V/div is indeed a good thing and should remain the default for scopes,  it would be a real help if it could be overridden.

I am pretty sure that no real-life bench-top scopes have a permanently-on automatic V/div setting. This is for the very good reason that there are times you don't want V/div changing randomly on when there are transients, or because you want to see some detail on only part of a signal not a full range deflection.   

While there are a number of people saying "Can I have this obscure component or that obscure component setting"  I suspect that those new obscure components, if provided, would remain unnoticed by most users. On the other hand I suspect nearly all users use the scopes ....

pfalstad, 2020-08-26:
I admit this would be very useful since I already have this feature (with no UI) and use it in several of the example circuits.

kesterlester, 2020-08-26:
Thank you. This would indeed be useful. Before posting in May I had a quick look at the source  to see if I could figure it out for myself, but found it beyond me. Nonetheless, if there is anything I can usefully help with (even if it is only usability testing for the UI) let me know. If there is anything I can assist with I will help...

pfalstad, 2020-09-10:
Added

closed 2020-09-10 by pfalstad