sharpie7/circuitjs1 #309: How to add a custom designed input signal (reading from file)?

E3V3A, 2019-02-28:
How can I add a custom designed data-driven input signal?

I would like to read voltages from a file and either repeat the signal at the end or just once. Obviously one would need to somehow set the time step constant between of each reading. This is real data coming from an EEG that I would like to read and filter. I understand that such a data file could become too large for the app to handle,  for long runs, but I only need to simulate 1-3 seconds with millisecond precision.

Not sure if this is already available somehow, but I could not find it. I'd be willing to try to create a new input device with this function if I could get some guidance how to best go about it. Thanks again for this  great simulation tool.

E3V3A, 2019-03-07:
I propose the new element as something between a voltage source and a `reversed` *DataRecorder*.

I particular I was looking at "[DataRecorderElm.java](https://github.com/sharpie7/circuitjs1/blob/master/src/com/lushprojects/circuitjs1/client/DataRecorderElm.java) and perhaps use that as a model for creating a **`DataInjectorElm.java`** with the following.

Features:
* The Data Injector circuit element should be enabled/disabled by a single click (like a switch)
* Inject the Voltage data from the selected file
* First line contain the *optional* `# time step = 0.000005 sec` describing the dT between each data point.  
  (If not present just use the simulation dT.)
* Each line is in the unit of [Volt]
* Available fields that can be edited:
```
    File Name:  [indata-20190307-1310.txt]
    ----------------------------------------------
    [ ] Use Single Shot (default: Repeating)
    [ ] Force Time Step (seconds):      [5u]      (to override what's in the file)
    [ ] Force Maximum Measurements (before signal Ends or Repeats)
        (*) In Seconds:                 [500m]
        ( ) In Number of data points:   [1M]
    [ ] Use Data Window with points:    [12-3456]
```

pfalstad, 2019-03-07:
The only trick here is how to handle loading/saving of the circuit containing this element.  I'd like to avoid having the entire data file saved in the circuit.  I don't think we can save a reference to the file and then load it again next time..(?)  So you would have to reload the file every time you load the circuit in order to use it.

Otherwise we could allow the user to specify a url instead of a file, but then you run into site cross-scripting issues.

E3V3A, 2019-03-07:
Very good point. But at the other hand, I think this would be quite a fringe usage case, although very useful for testing and playing with (weird) signals. So that understood, it should be ok that it doesn't load the *original* file. So what are our options?

1. Load nothing, but present the user with a *info/warning* box saying that he has to load a data file.
2. Load a template data file, telling user (in right hand side area) that it has loaded a template file and not the original or something else. Possibly with a URL to where it can be found. I guess this would also require that you specify an external URL in the Edit menu, that will be saved with the component.
3. Automatic Save/Reload the URL to a file hosted on any of the many *"pastebin"*-like sites
4. Automatic Save/Reload the URL to a Dropbox. (These tend to disappear!) 

That said we could also attempt something extraordinary, either by using [itty bitty](https://github.com/alcor/itty-bitty/) to save the circuit in the URL itself. [Most browsers](https://stackoverflow.com/questions/417142/what-is-the-maximum-length-of-a-url-in-different-browsers) can handle several KB's of data.

We can also reduce the data significantly by: 
* Compressing the data using `gzip -9` gives about 1/2 file size.
* Reducing the number of significant digits in the voltages, which are currently >16. Perhaps make it more sensible around ~6, perhaps with the addition of a `# voltage range: mV` in the header.

Those steps above would allow us to save ~7500 data point in 22 kB. (From 145 kB)

pfalstad, 2019-03-07:
I vote (1)..  3 and 4 have the same cross-site scripting issues.  The itty bitty solution is basically saving the entire data file in the circuit, only compressed.

pfalstad, 2020-12-13:
Use an Audio Input if you can encode your data as a .wav.

closed 2022-01-15 by pfalstad

E3V3A, 2022-01-23:
@pfalstad 
This is super cool! Thank you so much Paul!