sharpie7/circuitjs1 #313: Feature: Show power-frequency spectrum

E3V3A, 2019-03-06:
I came across [this](http://www.falstad.com/circuit/e-notch.html) older circuitjs version, of the `notch-filter` and found this very nice power-spectrum:
 
[image]

How can this scope presentation be done in current version?

pfalstad, 2019-03-06:
It's not an older version, it's a separate applet.  Go here and import your circuit:

http://www.falstad.com/afilter/

It's a separate applet because it doesn't support as many elements, the UI is different, and the simulation approach is completely different.

E3V3A, 2019-03-06:
Aha, I see. That is a very cool applet.
Would have been very nice to be able to see that type of plot, also in circuitjs. 
Is there any way that the plot engine could be beaten into something like that?

> he simulation approach is completely different

What approach is that?

closed 2019-03-06 by E3V3A