sharpie7/circuitjs1 #314: Confusing development workflow

E3V3A, 2019-03-11:
@pfalstad  @sharpie7 
Hi Guys,

I find your development workflow a bit confusing. One the one hand, Paul is making more continuous updates, merges etc. but without any bug issues available, while here at Iain's repo you do the bug issues and occasional PR merges. Now, this may be a great solution for you guys, it makes it hell for other people to contribute, maintain and fork from 2 different places, most often out of sync. Forcing you to cherry pick commits/PR's to stay up-to-date, while referring to bugs/issues from another repo.

To improve and simplify the situation, I would suggest you guys start a new github project/organization where you can combine all this in one place. I think this would be more beneficial and easier to maintain  for both of you, in the long run. 

What are your thoughts?

Cheers,

emilfihlman, 2019-03-15:
Agreed, the current situation is strange.

E3V3A, 2019-03-15:
### Where should I send my PR's?  
(They are based on Pauls repo...)

closed 2019-03-17 by sharpie7

sharpie7, 2019-03-17:
ETA:

The process works for us and has done for several years.

This version tends to be the master.

You could send me PRs. I did take a look at your fork. From what I saw, your priorities and preferences are very different from mine (and probably Paul's) so it's unlikely that I will incorporate changes of that nature in to this branch. In particular I would advise that I am not going to change any of the tooling or process unless Paul and I are completely bought in to it.

I have some sympathy with your proposal to address the default ranges for the sliders. If you wanted to make code changes to address that with in the existing code and tooling then that is the kind of thing that would be useful.

E3V3A, 2019-03-17:
@sharpie7 
> your priorities and preferences are very different from mine

I'm not sure what you are referring to there?

> I am not going to change any of the tooling or process unless Paul and I are completely bought in to it.

I didn't think so, and that is fine. However, as I mentioned in #315, I would suggest to at least include the 3 files (*`ecl2mvn.sh, pom.xml` and `build.gradle`*) to so that people can easily convert to use whatever they deem fit.