sharpie7/circuitjs1 #315: Automatic conversion to Maven or Gradle builds

E3V3A, 2019-03-14:
Hi Guys!

I have made a Bash script and gradle build script, that will automatically convert your Eclipse based source paths into a Maven or Gradle based build environment. I still need to do some tests before I hand it over, but you can expect this shortly. So I will provide:

1. a Bash path conversion script
2. an updated and corrected `pom.xml` (for **Maven** builds)
3. a working **`build.gradle`** script that will build entire site (for **Gradle** builds)

Let me know if you're interested.

E3V3A, 2019-03-14:
Done!

If you already have Maven or Gradle installed, you can now build your site with 2-3 lines:
```bash
./ecl2mvn.sh
# mvn install
gradle compileGwt
gradle makeSite
```

:tada:  :balloon:  :+1:

closed 2019-03-17 by sharpie7

pfalstad, 2024-10-20:
Thanks!  I incorporated your build.gradle into my repository (finally).