sharpie7/circuitjs1 #317: [Feature Request] Picture import/overlay

sle118, 2019-03-25:
I am suggesting a feature which would simplify the reproduction of existing circuits for simulation.  

Having a full blown importer that would be compatible with circuit design software would be amazing but probably require a considerable amount of effort to achieve.   So being able to take a static picture of a circuit design to draw on top of and place component would speed the manual process up.

thank you!

E3V3A, 2019-04-03:
@sle118 
>  So being able to take a static picture of a circuit design to draw on top of and place component 

TBH, I have no idea what you are talking about! Sound like a unicorn to me. So please describe in details what you want to do, and if this is something somebody can do in a few evenings, or if we need 3 years of 24/7 development.

* What do you mean "taking a static picture"?
* What do you mean "draw on top"?
* What do you mean "place component on top"?

pfalstad, 2019-04-03:
I think he means he wants to be able to import an image into the simulator.  The image would be in the background and serve as a guide so he can place components on top of it.

sle118, 2019-04-03:
I'm sorry I wasn't clear enough to you!

@pfalstad  is spot on.

Some background context: I'm troubleshooting an amplifier power supply circuit and would like to position the components as close as possible to their physical layout on the PCB.  

This feature would allow:
1- Grab a screenshot of the PCB layout from the service manual
2- Import it into the simulator, which would draw it as an image "component" (that could be resized/moved)
3- Users would start inserting components and move/scale them, using the image component as a reference
4- Users can add different measurement points to the circuit, print the resulting simulation, probe the real circuit with an oscilloscope to compare with the expected results

I understand the limitations of a simulator and that there is a likelyhood that its results won't match reality, but at least it provides a way to get some insights on how a circuit is supposed to work. 

Let me know if this makes it clearer.  Thanks!

E3V3A, 2019-04-03:
Oh yeah, now that I understand what you meant, it is an excellent idea! :1st_place_medal: