sharpie7/circuitjs1 #321: Feature Request: Capacitor and Inductor ESR values

K4czu5, 2019-04-08:
Could you add ESR factor to capacitor and inductor?  It would be helpful in more advanced schematics where placing resistor and other component is inconvenient.

Thanks in advance.

pfalstad, 2019-04-09:
Equivalent series resistance?  That probably wouldn't be that hard.  Adding a small default series resistance to all capacitors and inductors might solve some other issues, too (like capacitor loops, etc).

K4czu5, 2019-04-10:
Yes, that's it. Thank you, very good work, keep it up

Exaeta, 2019-04-30:
Please add a few different capacitor parameters:

Type:

Aluminium Electrolytic
Tantalum
Ceramic
Polyester (aka Mylar)
Polypropylene

There is usually also a small series inductance and a large resistance across the plates, which varies with applied type. I find myself always adding these things manually.

ormaaj, 2019-06-05:
I'd guess @Exaeta is right and a bit of series inductance ought to be closer to reality. A capacitor being a big sheet of metal internally ought to have effectively zero DC resistance and the entire illusion of resistance has to be electromagnetic effects.

I know that a datasheet for most decent capacitors specifies the frequency for which an "ESR" is valid.

Exaeta, 2019-06-06:
@ormaaj The series inductance is a lot lower than you might think, but it is there. There's a lot more resistance.

esaruoho, 2026-03-03:
Fixed in v3-dev — capacitor series resistance added.