sharpie7/circuitjs1 #330: Feature Request: Realistic OP amps

Exaeta, 2019-04-30:
I would like to request more realistic OP amps..

Real op amps have a number of features that make the simulator less than realistic...
In real circuits, you usually need a bypass capacitor, because of the current draw from the + and - pins.
The following things would be helpful:

Option to enable + and - pins
Ability to set slew rate
Ability to set GBP
Ability to set common mode range
Ability to enable non-zero input bias current
Ability to set minimum and maximum output voltages relative to Vcc and Vee.
Ability to set maximum source and sink current on op amp output.

pfalstad, 2019-05-06:
https://github.com/sharpie7/circuitjs1/issues/331#issuecomment-489392306

ornaaj asked: "Is the 741 sufficient as a representative op-amp model? Yeah it's often used for illustration due to its historical significance and "simple" 3-stage bipolar design."

Good question.  I can easily adjust the output current and slew rate.  But the 741 isn't rail to rail.  Another op-amp model might be better, if it's simple.  The 741 is complicated enough as it is.

I found an LM324 model that is actually less complicated, and it gets closer to the rails, so maybe I'll use that?

Exaeta, 2019-05-06:
I was thinking a slew rate and relative rail limit would be a good improvement, since it allows you to get a semi-accurate estimate of what would happen for a wide variety of op amps. 

So how about a "select op amp" which has several, plus a custom option which allows you to set these parameters customly, instead of using a more realistic model.