sharpie7/circuitjs1 #331: Feature Request: Realistic Wire Option

Exaeta, 2019-04-30:
I would like to request a feature which enables a small resistance on each wire relative to it's length, as well as a small parasitic capacitance to ground. This would be disabled by default and the amount adjustable in options. So each wire would act as a resistor according to its length when the feature is enabled.

ormaaj, 2019-05-01:
The transmission line is sort of this, though not really a practical wire substitute presently.. Maybe if it had single-ended and differential variants, more flexible rotation, and less whining about its precious ground. :grin:

sharpie7, 2019-05-01:
Can you give an example of a circuit where this level of modelling is helpful? Baring in mind that this is not inteded as a tool for circuit design - it is more for illustration of circuit concepts.

There are a lot of factors to consdier if you really want to make a "realistic" model of a wire. They have inductive and capacitative effects to all other parts of the circuit.

ormaaj, 2019-05-05:
Is the 741 sufficient as a representative op-amp model? Yeah it's often used for illustration due to its historical significance and "simple" 3-stage bipolar design.

Sub-circuits would really be a killer feature.

Exaeta, 2019-05-06:
@sharpie7 I think it's useful to show how small capacitors in close proximity can help i.e. bypass capacitors.