sharpie7/circuitjs1 #332: Feature Request: TVS Diode, MOV

Exaeta, 2019-05-06:
I was wondering if we could get TVS diodes and MOVs?

You can (sort of) simulate it with two zeners and a resistor, but the voltage/current characteristics are a bit different and it's be nice to be able to add them as one component.

sandorex, 2021-03-07:
Sad this hasn't gotten any love for years

pfalstad, 2021-03-07:
It seems that you can simulate a unidirectional TVS diode by a zener, and a bidirectional TVS diode by 2 zeners.

You can do a MOV with a VCCS.  Here is a SiC MOV: https://tinyurl.com/ybtuqrk5  It seems to match the I-V curve shown here: https://eepower.com/resistor-guide/resistor-types/varistor/#

Here is a ZnO version: https://tinyurl.com/ybaw8977

pfalstad, 2021-03-07:
I guess he already said you could use 2 zeners to do a TVS.

pfalstad, 2023-01-09:
If people are happy with the VCCS-based MOV, I could a real component based on it.