sharpie7/circuitjs1 #333: MOSFET capacitance

SusiKette, 2019-05-06:
I noticed that the simulator doesn't simulate capacitance for MOSFETs. This would be very useful feature in my opinion. Building a capacitor mess around the MOSFET (similar to what "CMOS Inverter (w/capacitance)" example circuit uses) just to simulate capacitance takes additional time and looks a bit messy especially if you have more than two MOSFETs. I also noticed that in some cases N-type MOSFETs seem to let current through even though the gate isn't powered at all (similar bug sometimes seems to happen with some switches as well when they are open).

pfalstad, 2019-05-06:
Re the second part with the MOSFETs and switches letting current through, can you post an example?  use File->Export as Text or Export as Link.  Thanks

SusiKette, 2019-05-07:
The easiest way of seeing what I mean is to just change "Write" to 1 and "Read" will change to 1 too, regardless of X/SB begin 0.

Link:
http://tinyurl.com/y6ajaxem

This is also one circuit that would benefit from MOSFETs having capacitance built into them.

pfalstad, 2019-05-07:
Right, I've seen that before.  Not sure how to fix that.  When a MOSFET is off, we don't treat it as completely off, we instead treat it as a large resistor (10Meg).  If we didn't do that, we'd get a singular matrix error, because there would be no way to calculate the voltage of the "Read" node...  It's floating.

Since the "Read" node is connected to the other side of the MOSFET by a large resistor which has no current across it, both sides have the same voltage.  And the logic output senses voltage without requiring any current.

So to get around this, you need a pulldown resistor at the Read node (to force the node to ground unless there's actual current), or a capacitor to ground (so the voltage of the node can't change without current).

Exaeta, 2019-05-08:
@pfalstad I think what he is getting at is the "gate charge" i.e. C_OSS C_ISS etc.

For example, IRF540N has 1220pF C_ISS at 25V.
 
First you have the "off region", which is straight up, a near flat "turning on" region, and another linear "on" region. Setting the threshold turn on charge, off capacitance, turn-on capacitance, and on capacitance might model this approximately?

Exaeta, 2019-05-08:
Maybe something similar to this would be a slight improvement though, if not going for the realistic version:

http://tinyurl.com/y69w2dkv

SusiKette, 2019-05-10:
> Right, I've seen that before. Not sure how to fix that. When a MOSFET is off, we don't treat it as completely off, we instead treat it as a large resistor (10Meg). If we didn't do that, we'd get a [...]

Begin able to simulate a floating state probably should be included. There are circuits that rely on it. As far as I know, the voltage of a floating section is based on the charge. Since the capacitance is not simulated for MOSFETs there is no charge to discharge when a section is left floating and then there is no way to calculate the voltage.

A good example is the bus lines in 6502 CPU. It is a line that is not directly connected to a power source or ground. All connections are done with pre-charge, drain or pass MOSFETs. Pre-charge MOSFETs first charge all lines and then a pass MOSFET opens a path to a register, which selectively pulls down the bit lines to get the bus value to match the register's. The 6502 has plenty of examples where capacitance and floating state are useful.

ormaaj, 2019-06-26:
> Begin able to simulate a floating state probably should be included. There are circuits that rely on it. As far as I know, the voltage of a floating section is based on the charge. Since the [...]

It would seem that the FET gate is already "charge-controlled" in a way:
[image]

I *think* this successfully simulates a floating circuit connected to the gate. Note the lack of wire from the FET's source back to the gate supply. If that's true then the gate is doing as an ideal capacitance-free gate would.

[FET-bias.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/3327726/FET-bias.circuitjs.txt)

Perhaps simulating the body diode as a small varactor diode would do the trick? Then there would be effectively a gate-body capacitance.

gsuberland, 2022-02-06:
Adding my support for this feature. I've been manually simulating the capacitance along with gate resistance and inductance, but it's time consuming to set up and detracts from the clarity of the circuit. I don't know enough Java to implement it myself, otherwise I'd contribute it in a PR. I'd be happy to see it added either as a tickbox option to the MOSFET itself, or in a cloned version of the MOSFET in the same way that ideal and real opamps are delineated.

ormaaj, 2022-02-09:
The MOSFET model needs complete replacement really as has been discussed elsewhere. Not much point in changing this one because you can't really derive the geometry-dependent MOSFET gate and junction capacitances. They depend on the type of MOSFET. E.g. VDMOS VDS dependent capacitance can be added with an explicit explicit nonlinear capacitor as in this example https://seventransistorlabs.com/Modeling/SPICE/C_CJO.ckt

To add caps to the current model a subcircuit is the solution. Subcircuits just can't be mapped to another symbol yet and there is no way to pass parameters other than through pin currents and voltages. Those would be the requisite things to fix.

esaruoho, 2026-03-03:
Addressed in pfalstad/circuitjs1 PR #266 — adds parasitic junction capacitance (Cgs, Cgd, Cds) to MOSFET and (Cbe, Cbc) to BJT.