sharpie7/circuitjs1 #339: Bug - Creating Subcircuit Causes Javascript Errors

AndrewIsOffline, 2019-06-08:
Hello! I wanted to simulate a simple wall plug circuit, but one of the components, the LM78LXX voltage regulator, is not available in the program (it's kinda pointless anyways), so I recreated it using the equivalent circuit diagram in the chip's datasheet, and after tweaking out some ambiguities I was able to get it working. However, after I converted it into a subcircuit, it jammed up the "add subcircuit instance" tool, meaning that no subcircuits could be added. The specific error that it throws is "Exception caught: (TypeError) : Cannot read property 'Pe' of null" and the stack trace is:

En | circuitjs1-0.js:1926
-- | --
Qe | circuitjs1-0.js:504
xl | circuitjs1-0.js:1901
Se | circuitjs1-0.js:1702
af | circuitjs1-0.js:2276
DF | circuitjs1-0.js:1458
bH | circuitjs1-0.js:1345
Ig | circuitjs1-0.js:805
Lg | circuitjs1-0.js:1229
(anonymous) | circuitjs1-0.js:1745

Attached is the regulator circuit.
[circuit-20190607-2258.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/3268053/circuit-20190607-2258.circuitjs.txt)

pfalstad, 2019-06-08:
whoops, it looks like subcircuits containing mosfets or jfets don't work.  I'll fix it.

AndrewIsOffline, 2019-06-08:
Just tested it and everything works 100%.
You know, if there was an option to import/export subcircuits, this could allow people to easily share common ICs online. I know I wouldn't mind sharing some, like the LM78 (even though I'm not 100% sure about it...)

closed 2019-06-08 by AndrewIsOffline