sharpie7/circuitjs1 #342: Would it be easy to add an Audio Input?

justin-romano, 2019-06-12:
similar to the ac voltage sources so you could say upload a sample of your clean guitar sound and hear how a circuit would affect the signal. I think multisim has one.

sharpie7, 2019-07-07:
Fixed by @pfalstad

closed 2019-07-07 by sharpie7

justin-romano, 2019-07-08:
ha. I checked the source out and was going to do it myself. but did not get around to it. Just noticed the feature the other day and it occured to me that i had overlooked and it had always been there until i noticed the commit notification.
I tested it. works well.
Thanks Paul. (Love this `software BTW. its a fascinating way to express ideas in electronics and when you get the hang of it you can work with it really fast)