sharpie7/circuitjs1 #344: Can relays latch?

phazaar, 2019-06-25:
Forgive me if I'm being a Luddite here and missing a naming convention or an easy way around this, but I can't find any way to make a simple latching relay circuit. These are one of the most common components I tend to use, so before I go to the length of working through the code to create one, am I missing something obvious?

The behaviour I want is pretty simple: a relay that flips once when a current is applied, remains flipped when that current stops, then flips back to its original position when a current is applied the next time remains there when that current stops etc etc. I've had a look through other component options and the (impressive if confusing) 'toggle flip-flop' circuit example, but none actually do what this component ought to...

PS. Thanks for such a wonderful simulator; it's really saved my bacon on a number of occasions! :D

pfalstad, 2019-06-27:
No, those aren't supported.  Someone asked for them in #144.  It should be easy enough to make one with a subcircuit.

Weishaupt, 2022-09-01:
If possible could you provide an example of a subcircuit for latching relays? I wasn't able to come up with one myself.