sharpie7/circuitjs1 #349: Tinyurl link creation broken - July 2019

altimumdelta, 2019-07-06:
Maybe another one like bit.ly could be added too. Thanks

pfalstad, 2019-07-06:
seems like something is broken on the tinyurl side..  links with %0A in them can't be shortened using the API.  might have to switch to bit.ly or something else.

pfalstad, 2019-07-06:
try it now..  Looks like it was on our side.  I'm not sure how it was working before.

altimumdelta, 2019-07-06:
Originally It happened to someone else, I did my own test now and I saw the %0A being in there and it looks like it works for me in latest opera.

sharpie7, 2019-07-07:
Hi @pfalstad did you change the php script to fix the tinyurl problem? If so, could you upload the new verion please.

altimumdelta, 2019-07-07:
There were some people saying it didn't work for them as in the tinyurl link was created but was showing an incomplete diagram.

pfalstad, 2019-07-07:
Could you send us the circuits that were trying to create links for?

altimumdelta, 2019-07-07:
Oh I forgot, I can send the resulting tinyurl, but I don't have the source link, I can ask the person but I'm not  sure if they saved that one anywhere.

pfalstad, 2019-07-07:
having the tinyurl might help, yes.

altimumdelta, 2019-07-07:
https://tinyurl.com/y6sh7l64

pfalstad, 2019-07-07:
How'd you make the proper version?

pfalstad, 2019-07-07:
I'm not sure how the broken version happened but I think it's a different bug.  I can dump a proper version from tinyurl just fine: http://tinyurl.com/yxh3e8uh

altimumdelta, 2019-07-07:
Could be browser difference, the bug didn't happen to me so it's harder, it was happening to a bunch of people, could have been coincidences, bad luck. I'll be posting back if I find out more.

sharpie7, 2019-07-08:
Maybe I have a different problem, but I am just getting this error on my site: 
`<!DOCTYPE html> <html lang="en"> <head> <meta charset="utf-8"> <meta http-equiv="X-UA-Compatible" content="IE=edge"> <meta name="viewport" content="width=device-width, initial-scale=1"> <title>Error</title>  <link href="https://fonts.googleapis.com/css?family=Raleway:100,600" rel="stylesheet" type="text/css">  <style> html, body { background-color: #fff; color: #636b6f; font-family: 'Raleway', sans-serif; font-weight: 100; height: 100vh; margin: 0; } .full-height { height: 100vh; } .flex-center { align-items: center; display: flex; justify-content: center; } .position-ref { position: relative; } .content { text-align: center; } .title { font-size: 36px; padding: 20px; } </style> </head> <body> <div class="flex-center position-ref full-height"> <div class="content"> <div class="title"> Whoops, looks like something went wrong. </div> </div> </div> </body> </html> `

altimumdelta, 2019-07-08:
I got more info, it is working now, it was returning a HTML document body instead of the short URL, I guess the person manually created the first "broken" tinyurl but didn't copy/paste the entire URL correctly so it was as if it's half-broken, but this is just speculation.

pfalstad, 2019-07-08:
@sharpie7 , that is the problem I fixed.  Did you update with the shortrelay.php I checked in?

altimumdelta, 2019-07-09:
I have had another incident, this time me personally, I was doing my own thing, modifying some circuits I found online, and I noticed this "backward current" in some cases where all the amps/volts would go up really big numbers and jitter very fast, that usually didn't do any problems, I hit one example where it didn't make sense, so I exported that to a file, and created an URL, here's what happened, the circuit was incomplete in both the tinyurl and the saved file, the LEDs went missing, I was fidding with the one that remains, right clicking and doing another custom model called `spice-default-1` which doesn't exist in the saved/tinyurl version, so it's missing.

Now, that original circuit had `FWDROP=1.34` on all the LEDs as the custom model, but in this case they all shown up as if they had the `1N34 Germanium` model selected, HOWEVER the behavior of the circuit was using the `FWDROP=1.34` still in the background because I studied it and knew which volts/amps correspond to that amount of forward voltage, when I was done fiddling with one of the LEDs I set it back to `FWDROP=1.34V` for original state, setting it to Germanium made it show different values, so other leds, which are now missing were somehow bugged from the get go when the link was opened, possibly the tinyurl from far back (at least more than a year AFAIK) is correct but the new version of this sim may not be behaving well with it?

Here is the tinyurl with the missing LEDs: http://tinyurl.com/y4ydrjyo

pfalstad, 2019-07-09:
Do you have a link to the original circuit you were modifying?

pfalstad, 2019-07-09:
Ok I can reproduce it.. If I go to an old version and create a circuit with custom diodes like this:

$ 1 0.000005 10.20027730826997 53 5 50
v 512 416 512 128 0 1 40 5 0 0 0.5
w 512 128 656 128 0
w 656 128 656 192 0
d 656 192 720 256 1 1
d 656 320 720 256 1 1
d 592 256 656 192 1 1
d 592 256 656 320 1 1
w 656 320 656 416 0
w 656 416 512 416 0
w 592 256 592 352 0
w 720 256 768 256 0
w 592 352 768 352 0
r 768 256 768 352 0 100
x 815 312 852 315 0 20 load
o 0 64 0 35 10 0.05 0 -1
o 12 64 0 35 5 0.05 1 -1

then load that into the current version and duplicate one of the diodes, and delete the duplicate, all the remaining diodes are now germanium.  If I export that, the model description won't get dumped.  If I reload the simulator and try to import that file, the diodes won't get created.

pfalstad, 2020-08-02:
Looks like I fixed the diode export issue on 7/9/19, and the tinyurl problem on 9/12/19.

closed 2020-08-02 by pfalstad