sharpie7/circuitjs1 #355: Filter by extension in open file dialog

altimumdelta, 2019-07-09:
It may be handy to filter by the current .txt extension so that it ommits other files from view for faster browsing.

Speaking about extensions, I rather have a custom extension name for better separation, like .fal or fcsf (falstad circuit save file) because it's not like this is the only JS circuit sim right? could be mistaken for something else, but I kinda have a lot going on on my PC so maybe it's just me.

pfalstad, 2019-07-09:
Good idea to add a filter.  I picked .txt because I wanted to make sure the server and browser treated it like a text file, so it would have the correct line endings, etc.

pfalstad, 2019-07-10:
I added this but then backed it out.  I want to think about this some more.  I wonder if someone is using some other extension and will not be able to load their circuit files if I add a filter?

altimumdelta, 2019-07-10:
If some people rename their collection of circuitjs extensions? Not that I know of, but allright, maybe wait a bit to find out.

sharpie7, 2019-07-10:
> 
> 
> I added this but then backed it out. I want to think about this some more. I wonder if someone is using some other extension and will not be able to load their circuit files if I add a filter?

Yes, given we never really tried to have a standard extension for the exports I think the disadvantages outweigh the advantages of trying to introduce a filter now.

altimumdelta, 2019-07-10:
Actually the filter is one thing and another extension separate, it can filter just to .txt right?

sharpie7, 2019-07-10:
I quite often open files without a .txt extension, either because I changed it when I saved or it was edited in another application.

altimumdelta, 2019-07-10:
I guess you're not talking about windows?

pfalstad, 2019-07-11:
You can also just type *.txt in the file chooser, and it will filter by *.txt (on windows).

On mac, you can't easily switch to "all files".  You have to click an options button which is not obvious.  Also the filter doesn't really benefit you, since it just grays out anything that's not txt, rather than hiding it.

altimumdelta, 2019-07-11:
Sure. It can be achieved using the search box, but yes it does hide everything else on windows.