sharpie7/circuitjs1 #356: The justification of adding an IR LED item or model ?

altimumdelta, 2019-07-09:
I've started doing things with IR leds, with Wii Remotes and head tracking stuff, and more on the way in future project that I'm planning.

But I'm not that of an expert on them and the tech behind it, I thought of opening this for anyone who might add some pros/cons whether this would be worthwhile to add, if not anything at least for the coloration and adjusted properties dialog.

Another use case is when normal LEDs and IR LEDs are both used in the same circuit, the color of the IR LED could be kinda light-cyan-silver with an "IR" sign inside perhaps.

sharpie7, 2019-07-09:
From an electical point of view the existing LED model can be customized to match any IR LED. The colour on the screen (which isn't used in the simulation) can be set to any value you like.

altimumdelta, 2019-07-12:
However I think the whole LED GUI should be overhauled as this one is a bit awkward to be honest, the editing, naming and configuring the models is confusing and the GUI doesn't seem to work right.

I would first more visibly separate the esthetics from simulation configurations, enlarging the menu and creating 2 sections, the first being the main sim stuff and the esthetics such as color on the bottom.
This is just one possibility, nothing definitive tho.

The forward voltage namings should be normalized, either shold be all Vf or FV when using short versions IMO, "fwdrop" doesn't really look like a good abbreviation for it.

The built-in models should be separated and a checkbox added for "Use custom model". Custom models should have it's own area for editing without needing additional dialogs and those custom options should be expanded to accomodate the suggestion from some other people I'll reference below. A custom model shoul be filled out on the fly once the option is activated and the built-in section items greyed out; without creating a dialog at the top of the browser "Falstad.Com says "FWD Voltage at 1A"" which is yet another step and doing this for many LEDs and many trial-and-errof changes of the Forward Voltage and it becomes quite clicky. FWD Voltage could be made to be adjustable in a slider.

This would be the default custom model with the items being saved on fly as is, but if a person wants to make more custom models it could click a save button and it would add the new custom model to the Custom Model dropdown or similar and then it would ask for a name in a new dialog perhaps.

Other people say that currently this circuit for LEDs isn't as useful as it could be mainly because, I'm quoting, various LEDs aren't rated for 1A like this circuit assumes and because that's not adjustable it makes no sense and won't show correct numbers. The correct solution would be for the custom model dialog to allow you to specify the forward current that the forward voltage you're providing is at, and he goes on to say that it may take significant math for the circuit to define the custom model.

I did my own circuit and I had no specs on my IR LED diodes that I got from one particular retails store, and I tried to get it as accurate in voltage drop off a 100ohm resistor as possible, unfortunately it doesn't show voltage by two decimal places, I would first add support for that if possible. so this curit should at least like 1mA higher than the IR LED did in real life, because this circuit isn't correcting for the variations in various LEDs out there apparently.

altimumdelta, 2019-07-12:
It may be a limited model following a different Vf vs If curve, I recreated my setup and it appears it's off by 0.8 mA too much, and 360 mV too little.

http://tinyurl.com/y4og9u9t

Unless my DMM is not correct, it's a new mid-level DMM so I'm not convinced it's off by that amount, because I have 5V coming off the PSU and it's a really good one, computer PSUs are made to be super voltage stable so if I get 5.03V that should be correct for this high-end PSU, otherwise I would have probably already noticed computer issues if the 5V line would be 4.7 or 5.3V

altimumdelta, 2019-07-12:
Update: I digged more and found out a way to match my real world example, by using an Emission Coefficient of 4.102, now that's interesting, but this is some good amount of trial-and-error, is this really the most practical way and shouldn't need more to it?

http://tinyurl.com/y3tgdx7l

pfalstad, 2019-07-12:
Can you find a spice model for your LED?  That should give you the numbers you need.

altimumdelta, 2019-07-13:
My specific case is solved, so this is for the future cases and other cases. I don't have the model number of these LEDs in this case so specs are very limited. The point of this suggestion is to avoid having to deal with finding a spice model and the circuitjs can do it on it's own.

altimumdelta, 2019-07-19:
Sorry about speaking before knowing the whole program in depth, Test Point seems to show more accurate voltage, oh good.

closed 2020-12-12 by sharpie7