sharpie7/circuitjs1 #361: Bug: Glitching simulation on the path when severing connection with LEDs

altimumdelta, 2019-07-14:
So I managed to isolate this glitch, when the connection is severed my moving a LED then all the A/V labels on wires will start glitching some ridicolous low or high number/unit very fast as if the sim still half-thinks it is connected, might also be a source of the lag.

But when moving an actual **Wire** out of the way to disconnect a circuit, it more properly disconnects and doesn't produce a glitch.

pfalstad, 2020-12-11:
Could you send a sample circuit using File->Export As Link?

altimumdelta, 2020-12-14:
Sorry it may take a bit for me to recall that and find the saved example if I did it with this at all, I might have to recreate it if I can't find it in my archives, it's been some time.

pfalstad, 2020-12-14:
I should have asked you sooner..  Your message must have slipped through the cracks.

closed 2021-12-11 by pfalstad