sharpie7/circuitjs1 #364: Missing Examples

KAZ-SYED-ABUTHAHIR, 2019-07-21:
While Removing Examples from the Menu might be a good decision which induces user to wire up their own circuits in the simulators. From the Teachers point of view the Examples circuits were very much valuable and had been a great aid in teaching. Kindly put them back.

pfalstad, 2019-07-21:
They are still there..  are they missing for you?  All of them, or just some of them?

KAZ-SYED-ABUTHAHIR, 2019-07-22:
Dear Prof,
Thank You,
They're certainly missing in the link : https://www.falstad.com/circuit/. Kindly have a look. 
[image]

But They're found in the Link: http://lushprojects.com/circuitjs/circuitjs.html.

Sincerely,
KAZ

KAZ-SYED-ABUTHAHIR, 2019-07-22:
Dear Prof, 
Thank You!
The Examples are certainly missing from your link: https://www.falstad.com/circuit/
Kindly have a look at the snapshot: 
[image]
But I've found them in the link : http://lushprojects.com/circuitjs/circuitjs.html
Sincerely,
KAZ

pfalstad, 2019-07-22:
Hi, I'm not sure what's going on.  It's working for me.  Try clearing your cache and try again.

KAZ-SYED-ABUTHAHIR, 2019-07-23:
Well Thank You.

sharpie7, 2019-07-31:
Seems like all is well now!

closed 2019-07-31 by sharpie7

KAZ-SYED-ABUTHAHIR, 2019-07-31:
Thanks All is Well Now. Kindly Make Provision for Assigning Negative Values
for Resistance. Or Am I asking too much?

pfalstad, 2019-08-07:
What do you want negative resistance for?  I mean, it's easy enough to do, but it's not realistic..  Is there a use for it?

sharpie7, 2019-08-07:
Yes, not keen on adding this unless you can show me where to  buy one.

KAZ-SYED-ABUTHAHIR, 2019-08-07:
Oh! Dear Me ! I can't buy a negative resistance, but I suppose I can make
one like Paul Horowitz in his famous Negative Resistor Demo @ MIT.  I was
much excited by the concept when I read this in "The Art of Electronics".
If negative resistors exist one could certainly violate First Law of
Thermodynamics and make perpetual motion machines without any hunch. At
least you add a negative resistor sub-circuit which black boxes op-amps and
power sources to enthuse people. It's only a suggestion and You're the
master your own code/ But your software works nice and great for my
undergraduate students. Thank You !

pfalstad, 2019-08-07:
Yeah that circuit is one of the examples already: https://www.falstad.com/circuit/e-nic-r.html

KAZ-SYED-ABUTHAHIR, 2019-08-08:
Thank You for your response. I've overlooked this circuit.