sharpie7/circuitjs1 #365: MOSFET fixed threshold / no body effect

ormaaj, 2019-07-27:
What model describes MOSFET behaviour using a "beta" and a fixed threshold? The few CMOS design textbooks' introductory chapters I looked at always model threshold as a function of the channel dimensions and substrate bias (among other things but those seem to be the most significant parameters). I haven't found a model that approximates a realistic variable threshold like this given a beta parameter and base threshold.

Evidently some microprocessors use a dynamic body bias in places as a power-saving feature because subthreshold current is a function of the threshold voltage, which is a function of body-source voltage for a given Vds, so they can cut down on leakage in unused parts of a circuit.

pfalstad, 2019-08-07:
Hi, I got it from Pillage, "Electronic Circuit & System Simulation Methods", which got it from Muller and Kamins, "Device Electronics for Integrated Circuits".

It's described as the "quadratic model" here: https://ecee.colorado.edu/~bart/book/models.htm  .  That link doesn't mention beta but it seems to be the same model if you set beta = mu Cox (W/L).  It also discusses the variable depletion layer model, which has a varying threshold voltage.

Someone has asked for the EKV model (#21) to be provided as an option, since it's much more accurate and can model things like subthreshold conductance..  And it would also fix your issue as well.

ormaaj, 2021-03-26:
@pfalstad [EKV](https://github.com/ekv26/model) on github. :D

pfalstad, 2021-03-26:
Wow that's a lot of parameters!  I would have to do what ngspice does and run a converter to generate C from this verilog code, and then convert the C to java.  Because the verilog code does not seem to calculate the matrix elements that I need..  It only gives the currents.

ormaaj, 2021-03-27:
Yeah. I see some recognizable parameters from ngspice level 3 / 9 so it could be meant to be easy to translate models (similar philosophy to sgp -> vbic). Anything that requires sophisticated "parameter extraction" and knowledge of process parameters of course can be rather annoying.

If I were choosing a practical model to port to Java I'd probably start from the ngspice level 3 / 9 (I've yet to spot a behavioural difference between those. I think level 9 is meant for experimenting with modifications since people depend on stable behaviour for level 3).

The only confusing bit is the way that some parameters override others. E.g. I've never actually seen a model that explicitly supplies `kp` for instance so they must always depend on that to be derived.

pfalstad, 2021-03-28:
I could easily steal the mos level 3 code from spice.  I have already done that with the spice transistor code.  But is it sufficient to model subthreshold conductance?  I thought you needed EKV for that, based on what little I know about it.

ormaaj, 2021-03-28:
Hm I've not tested sub-threshold effects. I'd guess no. Or it might have some heuristic to guess generic subthreshold effects. But at least it gets body effect (what I was originally looking at back when I posted this).

EKV seems rather obscure. I only heard about it when you mentioned it here, and skimmed some of the old papers from the 90s, but no implementation was available until I noticed this one appeared in the latest spice manual.

Oh yeah... The interesting "analog reciprocal" example doesn't really work. It's super broken in spice because these "floating mirror" / "beta multiplier" patterns are really touchy with the more detailed models. I did a slightly more realistic one and forced the curve to roughly fit 1/x (with some DC offset) by using a weird valued depletion mosfet... Still not great. https://tinyurl.com/yjmf3tza

I noticed the BJT model... Very nice! :D

pfalstad, 2021-03-28:
This should probably be a separate issue, but..  The analog reciprocal was contributed a while ago (by Felthry).  He either designed it himself, or he got it from someplace he didn't remember.  I should probably just delete it, if I can't replace it with a real circuit that's actually been used.

pfalstad, 2021-03-28:
Right, I confused your MOSFET issue with another one, especially since you mentioned subthreshold current.