sharpie7/circuitjs1 #366: New Feature request - sensors

Penpiggy, 2019-07-31:
Can I request for simple sensors such as light dependent resistor, thermistors and other such device to be included? I presume that it functions in a similar way to a simple switch (an on/off state) with just a different skin on it.

Since I use it at a secondary school level for my technology classes, it will be helpful as it will proof less confusing for the students, than if I use a simple switch to simulate the component.

sharpie7, 2019-07-31:
A few people have made similar suggestions in the past, but I have always been rather against it because it is just "visual sugar" around resistors, switches etc. My assumption was always that people should be able to understand the abstraction.

Thanks for actually providing an explanation and a use-case where it might be useful.

At the moment I still have mixed-feelings about the idea, but I will give it more consideration.

pfalstad, 2021-12-13:
These have been added.

closed 2021-12-13 by pfalstad