sharpie7/circuitjs1 #368: Feature Request: Current Transformer

SgtSchultz71, 2019-08-12:
Hello! I was wondering if it would be possible to add CTs in the simulation. I work around many different kinds of low-voltage and medium-voltage breakers and it would be useful to simulate their realistic interaction with substation power systems.

pfalstad, 2019-08-12:
It seems like you could simulate a current transformer with a regular transformer, couldn't you?

sharpie7, 2019-08-12:
Plus, I would note that there is no claim or guarantee that the simulation is accurate or  suitable for any purpose, particularly not where they may be safety consequences.

altaic, 2019-08-13:
Even professionals look to learn and your tool is awesome for fiddling around to get there— learning is a lifelong process. Please don’t draw conclusions that sacrifice that or set expectations/focus to the same end. I’m sorry to have to object to sharpie’s comment, however thank you both (Paul and sharpie7) so much for the excellently available tool that you guys have developed and maintained.

SgtSchultz71, 2019-08-13:
@pfalstad I have tried this before but I could not figure out a way to just extract the current with causing a major voltage drop messing up the rest of the circuit.

@sharpie7 I completely understand your disclaimer and I should note that I do not use this program to simulate work-related projects. I simply work around this stuff everyday and I was curious to learn and simulate the inner workings.

pfalstad, 2019-08-13:
I want to second what sharpie7 said...  There are nonlinear and other effects that the simulator doesn't take into account, and they could cause dangerous conditions.

This seems to match what I learned about current transformers from reading wikipedia: http://tinyurl.com/y3qr9v4l

You just need a transformer with a big ratio and a very small primary inductance (since the primary is just a wire going through a ring).

SgtSchultz71, 2019-08-14:
@pfalstad Thank you sir! That appears to have solved my problem.

closed 2019-08-14 by SgtSchultz71