sharpie7/circuitjs1 #372: feature request: more realistic Relay

demon740, 2019-08-19:
relays have two thresholds for turning on and remaining turned on. 
actually they have a hysteresis behavior.
for example they will be turned on with 100 ma  but they will remain turned on until 80 ma.

This circuit is automatic battery charger that works will real parts but you can not test it by this simulator:

$ 1 0.000005 10.634267539816555 43 2 50
g 352 304 352 352 0
t 304 256 352 256 0 1 -14.960958924875191 0.6685070695143888 100
v 48 240 48 144 0 0 40 16 0 0 0.5
g 48 304 48 352 0
r 176 80 304 80 0 3.3
178 384 80 448 80 0 1 0.2 0.015612679982862754 0.05 1000000 0.02 20
209 624 160 624 304 0 0.01 10.732079920213396 1
r 624 96 624 160 0 1
w 448 64 624 64 0
w 624 64 624 96 0
w 384 128 352 128 0
w 352 128 352 176 0
d 304 80 384 80 2 default
174 144 256 208 272 1 1000 0.797 Resistance
w 240 256 304 256 0
w 352 240 352 176 0
w 352 272 352 304 0
w 624 64 752 64 0
w 752 64 752 416 0
w 752 416 240 416 0
r 240 416 240 336 0 10000
w 240 336 240 256 0
g 144 256 144 352 0
g 624 304 624 352 0
d 432 288 432 224 2 default
162 448 96 528 96 2 default-led 0 1 0 0.01
r 528 96 528 304 0 1000
g 528 304 528 352 0
w 176 272 176 288 0
w 176 288 208 288 0
w 208 288 208 256 0
w 208 256 240 256 0
w 176 80 48 80 0
w 48 80 48 144 0
w 48 240 48 304 0
w 352 176 432 176 0
w 432 176 432 224 0
w 432 288 432 304 0
w 432 304 352 304 0
w 304 80 304 112 0
w 304 112 384 112 0
209 304 256 304 336 0 0.00001 0.6685070695143888 1
g 304 336 304 352 0
r 304 112 304 176 0 270
w 304 176 352 176 0
o 14 64 0 4099 1.25 0.000390625 0 2 14 3
o 6 64 0 4099 20 6.4 1 2 6 3

pfalstad, 2019-09-26:
Ok I made a change to add hysteresis and this circuit still didn't work.  I don't think that's the issue.  It seems that as soon as the relay starts to turn on, the transistor turns off and the current to the relay coil stops increasing.  So the current never gets high enough for the relay to switch all the way.

Right now, as the current approaches the "on current" value, it transitions from the off to on state as the current increases.  If the current is close to the "on current" but not quite high enough, it will be stuck in an intermediate state. 

I tried changing the relay so it has no intermediate state..  Once the current reaches the threshold, it switches immediately.  Then the circuit worked.  But having the relay switch instantaneously isn't realistic either.

Maybe the relay should switch after a predefined delay?  Once the relay starts to switch it doesn't stop, but it takes a configurable amount of time.

demon740, 2019-09-28:
Thank you for taking care of this matter.
actually  a relay is a electro-mechanical device and it's real behavior could be very complicated and it may not be the same for different type of relays.but as I know , the hysteresis  and some kind of delay, exist for all types of relays (with different values for thresholds and delays).I accept that my circuit will not work with all types of relays but I think in all of them, there is some delay between triggering them and changing there states and my circuit will not for with fast relays.
there is a simple circuit that works for all type of relays but it does not work in this simulator:
$ 1 0.000005 10.20027730826997 50 5 50178 240 176 384 176 0 1 0.2 0.009835088991201389 0.05 1000000 0.02 20g 240 288 240 304 0R 240 176 240 112 0 0 40 5 0 0 0.5r 240 224 240 288 0 100w 384 160 384 64 0w 384 64 176 64 0w 176 64 176 208 0w 176 208 240 208 0w 384 192 480 192 2o 8 64 0 4098 5 0.1 0 2 8 3

it is just a relay that it's trigger current passes from it NC pin.In all kind of relays that I have seen, you must have a square waveform in NO pin of relay, but in your simulator it behaves strangely.

pfalstad, 2019-09-28:
I can't read that circuit..  It had all the line breaks removed.  To post a circuit you can get a link to it from File->Export Link, then click "Create Short URL" to make a short link you can easily post.

demon740, 2019-09-28:
sorry about that.
http://www.falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+10.20027730826997+50+5+50%0A178+240+176+384+176+0+1+0.2+0.004358704489023573+0.05+1000000+0.02+20%0Ag+240+288+240+304+0%0AR+240+176+240+112+0+0+40+5+0+0+0.5%0Ar+240+224+240+288+0+100%0Aw+384+160+464+160+0%0Aw+464+160+464+352+0%0Aw+464+352+192+352+0%0Aw+192+352+192+208+0%0Aw+192+208+240+208+0%0Aw+384+192+432+192+0%0Ao+9+64+0+4098+5+0.1+0+2+9+3%0A

and here is the short url:http://tinyurl.com/y6ogrv3q

pfalstad, 2019-09-28:
Oh yeah, I remember making one of those once.  So it's supposed to oscillate?

demon740, 2019-09-28:
yes it must oscillate and its freq depends on physical structure of the relay but its about 10-60 Hz.

pfalstad, 2019-10-03:
I don't suppose you have ever hooked up one of these circuits to a scope?  I'm wondering what the duty cycle is like.  I'm working on this, and the relay immediately switches off once it switches on, making it a pretty low duty cycle.  I'm wondering if there needs to be another delay that determines how long the relay stays in its new state before switching back?

demon740, 2019-10-05:
yes I did not connect it to scope really and I think the duty will be dependent to structure of relay (may be less or more than 50%). there is two delay: 1- duration between triggering a relay and changing the state and 2- duration between disconnection of power of a relay and changing the statesthese delays completely depend on physical structure of relay. 
you can imagine a simple model for a relay as below:
Connection scenario:1- power is connected to coil ( voltage becomes more than  V_on_threshold)2- delay_on_13- NC pin is disconnected from Common pin4- delay_on_25- NO pin is connected to Common pin
Disconnection scenario1- power is disconnected from coil ( voltage becomes less than V_off_threshold)2- delay_off_13- NO pin is disconnected from Common pin4- delay_off_25- NC pin is connected to Common pin

(V_on_threshold > V_off_threshold)

as you can see, in each transition, there are situations that Common pin is not connected to NC or NO.
and the duty cycle depends to these values: delay_on_1 , delay_on_2 , delay_off_1 , delay_off_2