sharpie7/circuitjs1 #373: Tri-state buffer intensity

SgtSchultz71, 2019-08-22:
Why is it that the tr-state buffer seems so resource intensive? I am only able to place a few of them before I need to drastically slow down my simulation. Just a curiosity.

pfalstad, 2019-08-22:
How many are we talking about?  I created a circuit with 8 of them and it didn't seem that bad.  I could run it at about 80% full speed.

The tristate is going to be a lot more resource intensive than logic gates, maybe the equivalent of 3-4 inverters?  We can't do a bunch of optimizations that are easy to do on most digital components.

SgtSchultz71, 2019-08-23:
Well somtimes I try to simulate abstracted computer components that involve RAM, so I use many (8+) tri-state buffers to simulate a Read operation to a bus. I've learned some good tricks to overcome the issue though like using a DAC to a buffer and then to a ADC. I was just wondering.