sharpie7/circuitjs1 #374: [Feature Request] AVR simulation

nixie84, 2019-08-23:
Is there any chance to add AVR simulation?
https://github.com/buserror/simavr
With this feature it would be the best simulator ever!

ormaaj, 2019-08-26:
No chance if it can't run in a browser. Nearly no chance otherwise.

billcollis, 2019-08-26:
Nearly is not quite accurate,
I have an drag and drop AVR and ESP32 simulator I built, which is part of a suite of tools I use with my engineering students at the University of Auckland.
please feel free to check it out at https://xplainittome.azurewebsites.net/AVRQuestionPresenter/

JonasForssell, 2020-07-21:
I second this feature request. Having a CPU component would be very powerful addition indeed.
To not complicate the simulator applet too much, the component could just have a link to an external file with the source code.
Program could then be written in an external editor.