sharpie7/circuitjs1 #375: Feature required: Template to create new electrical and electromechanical devices

rricon, 2019-08-24:
The idea is to be able to complete a template with data from a new device. That is, each user can create a new device without having to know the details of the programming. For example, an automotive switch is a set of mechanically controlled electrical switches in an allowed order.

pfalstad, 2019-08-24:
You could use either a subcircuit or custom logic, depending on the specifics.

rricon, 2019-08-24:
Yes, it is correct, but I mean introducing electromechanical devices such as an electric motor. In addition, the idea of the template is to allow the community to participate in increasing the library of the simulator.

pfalstad, 2019-08-24:
Ok, I understand now.  Yes, this would be good.

SgtSchultz71, 2019-08-24:
It would be awesome if we could share custom logic designs so that way people can share the best version of a particular custom logic block. An example would be like a program counter or a RAM module. A kind of library to pull other people's collaborations or works from.

rricon, 2019-08-24:
That is the key word, collaboration. We are an army of users around the world that we can collaborate designing new devices. To achieve the goal, there must be something similar to a template that is an interface between the searched model and the javascript code.

ormaaj, 2019-08-26:
At some point it would become necessary to replace the symbol graphic definitions with support for reading kicad symbol files, and allow mapping them to subcircuits.

sharpie7, 2019-08-26:
Anyone can collaborate already by changing the Java code to create new components. It is a bit hard, but quite a few different people have contributed component models that are in the simulation today so it can be done.

In principle it is a good idea to create more a "plug in" style environment for new components and symbols, but it is a lot of work. There is sometimes "black magic" in writing the model for a component so it actually works in the simulation. It isn't always a direct implementation of the maths that you might expect.

Liniking the electrical simulation to energy paths outside the electrical domain brings an additional set of questions in terms of how you can best model and display the interaction between electrical and other forces.

sharpie7, 2019-08-26:
To ammend my previous comment a little bit: one of the main problems with contributing directly to the code is that the model needs to meet certain criteria if we are going to include it in the main branch. For example, it needs to have a certain level of quality and it needs to have a certain level of general interest.

If it was possible to edit models in user space then people could just use and share their own models (good or bad) without any cooperation from the application developers. Some people might find that useful, particularly if they have unusual requirements. The downside is that if people start distributing examples with bad models then that could negatively affect the perception of the project.

@rricon It's an open source project - pretty much each component is a java file named after its type. You can pick something similar and start editing.

sharpie7, 2019-08-26:
@rricon I understand this is more effort than you are willing to put in. I am just pointing out that user created components are already possible, and that, even with a better interface, they are always going to require quite a lot of user effort IMHO. 

You haven't really explained your use case. What would you actually do if this feature was available?

eMPee584, 2020-11-24:
@rricon have you heard about OpenModelica? It's a cross-domain modelling / simulation framework..