sharpie7/circuitjs1 #378: Transformer shows extra big voltages when 1 of its Coil is disconnected (Trapezoidal mode)

dr-vij, 2019-08-28:
Hello, I've found a situation, if you take a transformer, connect it to source, and then disconnect it from source - it starts showing extra big boltages with different sign at each step. I see this bug for trapezoidal method. When back euler - everything is ok.
[image]
[image]

pfalstad, 2019-08-28:
Yes..  use backward Euler for transformers in combination with switches or sudden changes in current, to avoid this problem.

SgtSchultz71, 2019-08-29:
Isn't that just the voltage spike associated with inductors when they lose voltage and the field collapses backwards?

pfalstad, 2019-08-29:
No, since this is an ideal switch, it should stop the flow of current immediately.  If it were a non-ideal switch, there would be some capacitance across it and there would be kickback, but it wouldn't last forever like this does.

dr-vij, 2019-09-02:
Thank you, now I understand

closed 2020-01-02 by sharpie7