sharpie7/circuitjs1 #380: Assumed grounds not shown?

terminator356, 2019-09-10:
Hello. I noticed that for example in all of the transformer examples,
 the bottom wires seem to be assumed at ground but there is no ground shown.
It seems to me that alternating voltage should be shown on the bottom wires
 of the transformer primary, and most certainly the secondary.
A student might get the wrong impression here.
Am I missing something? Thanks.
[image]

pfalstad, 2019-09-10:
yes, that makes sense.  we should add grounds to these.

terminator356, 2019-09-10:
Hm, I noticed that the same thing happens with a blank canvas and simple AC source.
It seems that both wires should have alternating voltage.
Or else show the assumed ground on the lower wire.
Perhaps it is a simple error in the code, that this two-terminal AC source thinks
that it is a one-terminal source, which is supposed have an assumed ground.
[image]

pfalstad, 2019-09-10:
If there is no ground, the simulator picks a node to be ground.  You should add a ground if you don't like the simulator's choice.  I don't see a need for the simulator to show the assumed ground in user-created circuits.

If you want a split supply, use two AC sources: http://tinyurl.com/y5qfktdp

terminator356, 2019-09-10:
Understood. Thanks.

pfalstad, 2020-12-11:
Added grounds to transformer examples.

closed 2020-12-11 by pfalstad