sharpie7/circuitjs1 #387: Showing voltages in percentage, as relative to some node, rather than absolute, plus color on wires based on percentage value

michaelsjackson, 2019-09-22:
The voltages could be shown as a relative value, in comparison against a selected node value, then having percentage values between 0..100, those numbers could be displayed in real-time as colors on the wires, plus on the circuits elements themselves.

Advantage would be you would see the relationships among values more precisely and visually. Just an idea, thanks for reading.

pfalstad, 2019-09-26:
Hmm, seems like a pretty specialized feature.  Instead, you could calculate a relative voltage (percentage value) using a VCVS, and then that would be displayed using the standard colors..  And you can adjust the range of the standard voltage colors from the options.

michaelsjackson, 2019-09-26:
Thanks for replying, standard colors I know only red, green, grey, what I meant was more mapping rgb colors to the range of 0..100, and the number meanings could be relative voltage values compared to some node voltage. Never tried any VCVS and its settings, maybe this is already possible, I will check again, thanks.

michaelsjackson, 2019-09-28:
But I also meant relative values on all wires, against a relative node, not only from a single wire which would be the above suggestion with VCVS.

closed 2019-11-23 by sharpie7