sharpie7/circuitjs1 #393: Offline Version for Linux(Mint 2019)

SM-Fahim, 2019-11-11:
In the home page, you've provided a file for offline version of the Circuit Simulator for Mac and Windows, but why not for Linux? I'm a Linux user Mechatronics Engineering student, and I like your software too much! But I badly need the offline version of it! Looking for your kind consideration.

pfalstad, 2019-11-12:
hi, try this and let me know if it works: http://www.falstad.com/circuit/circuitjs1-linux.tgz

SM-Fahim, 2019-11-14:
Thank you sir for you reply. But after extracting the file, I didn't find
any README file, so proceeded with the ./configure command, but it returned
"No such file or directory found". I tried to solve this, but couldn't make
it. Then I heard that you've a Java version which runs on Linux. So I think
I should go for that. But if you help me out about installing the one you
sent me, I'd love to install that.

pfalstad, 2019-11-14:
run the file circuitjs1: ./circuitjs1

SM-Fahim, 2019-11-14:
sorry, the java version didn't work. It got installed, but shows "Can't
read setup.txt!" . So I actually need your help about how to install the
software you sent me after extracting the file.

pfalstad, 2019-11-14:
for the one I sent you, try running circuitjs1

for the java version, how are you running it?  you should be able to run "java -jar circuit.jar" from the command line.

SM-Fahim, 2019-11-14:
Hello sir. This is the output:
fahim@Fahim:~/Downloads/Softwares/Fastad/circuitjs1$ ./circuitjs1
./circuitjs1: error while loading shared libraries: libXtst.so.6: cannot
open shared object file: No such file or directory

pfalstad, 2019-11-14:
hmm, you could try "sudo apt install libxt6:i386"

SM-Fahim, 2019-11-14:
After installing that, still the same result.
fahim@Fahim:~/Downloads/Softwares/Fastad/circuitjs1$ ./circuitjs1
./circuitjs1: error while loading shared libraries: libXtst.so.6: cannot
open shared object file: No such file or directory

rodolforg, 2019-11-14:
It should be `libxtst6:i386`.
I needed as well:

- libnss3

- libgdk-pixbuf2.0-0

- libgtk-3-0

So:
`sudo apt-get install libxtst6:i386 libnss3:i386 libgdk-pixbuf2.0-0:i386 libgtk-3-0:i386`
should do the work.

For running, I needed to do something strange. I couldn't run as a regular user, but this way:
`sudo ./circuitjs1 --no-sandbox`

pfalstad, 2019-11-15:
That is very strange.  What happens if you run without sudo, or without —no-sandbox?

rodolforg, 2019-11-15:
Oh. It works without `sudo` indeed. ;)

pfalstad, 2019-11-15:
Great, thanks for the info!

MakeMike, 2019-11-15:
Hey guys i am a new Linux user. Can you guys tell me the complete solved command for solving the issue.

SM-Fahim, 2019-11-16:
iyehhee!! It worked!!!
@rodolforg: Your command installed and setup something. Then I typed
./circuitjs1 and it worked! Hope so it functions well!
Additionally, after commanding ./circuitjs1, I got this output. Is this
okay?
fahim@Fahim:~/Downloads/Softwares/Fastad/circuitjs1$ ./circuitjs1
GLib-GIO-Message: 03:10:46.043: Using the 'memory' GSettings backend.  Your
settings will not be saved or shared with other applications.
[5734:1117/031055.423147:ERROR:buffer_manager.cc(488)]
[.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : glBufferData: <- error
from previous GL command

tom2238, 2019-11-17:
Hi. Thanks. It worked in linux Porteus 3.2.2 32bit.
Only download, extract and run:
sudo chown root:root chrome-sandbox
sudo chmod 4755 chrome-sandbox
./circuitjs1
Enjoy this offline linux version. Some package would be good.

closed 2019-11-18 by SM-Fahim

gskrish2001, 2020-03-14:
Hi Paul, thank you for your help, need some more pointers.
The download and the first two commands (tom's) worked:
(base) kg@kg-Latitude-E7450:~/Downloads/circuitjs1$ sudo chown root:root chrome-sandbox
[sudo] password for kg:  
(base) kg@kg-Latitude-E7450:~/Downloads/circuitjs1$ sudo chmod 4755 chrome-sandbox

The third command did not:
(base) kg@kg-Latitude-E7450:~/Downloads/circuitjs1$ ./circuitjs1
./circuitjs1: error while loading shared libraries: libX11-xcb.so.1: cannot open shared object file: No such file or directory

So I surmised I needed this library (I run Linux mint NOT Ubuntu), so I tried to install libX11-xcb-dev; again failure, I guess I do not know where to download it from.
(base) kg@kg-Latitude-E7450:~/Downloads/circuitjs1$ sudo apt-get install libX11-xcb-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libX11-xcb-dev

I did find a library for linux mint but not sure if it is the right one. Can you point me to the right one please. Thank you.

gskrish2001, 2020-03-14:
2nd update:
Installed libX11-xcb-dev from this website:
https://community.linuxmint.com/software/view/libx11-xcb-dev
Then tried tom's commands, still the same error.

pfalstad, 2020-03-14:
try: sudo apt-get install libx11-xcb1

xcb-dev is the development headers, which won't help you since it's already compiled.

gskrish2001, 2020-03-14:
**Hi Paul,
Guess this did not work. any other ideas**
(base) kg@kg-Latitude-E7450:~/Downloads/circuitjs1$ sudo apt-get install libx11-xcb1
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libx11-xcb1 is already the newest version (2:1.6.4-3ubuntu0.2).
0 upgraded, 0 newly installed, 0 to remove and 460 not upgraded.

**0 newly installed because I installed it just before and it installed.**

(base) kg@kg-Latitude-E7450:~/Downloads/circuitjs1$ sudo chown root:root chrome-sandbox
(base) kg@kg-Latitude-E7450:~/Downloads/circuitjs1$ sudo chmod 4755 chrome-sandbox
(base) kg@kg-Latitude-E7450:~/Downloads/circuitjs1$ ./circuitjs1
./circuitjs1: error while loading shared libraries: libX11-xcb.so.1: cannot open shared object file: No such file or directory
(base) kg@kg-Latitude-E7450:~/Downloads/circuitjs1$

gskrish2001, 2020-03-14:
This is a 64bit computer, could that be the problem? Thank you?

pfalstad, 2020-03-14:
You could try this: npm install electron -g

gskrish2001, 2020-03-15:
First installed npm. Then tried installing electron -g (gave errors); so I installed in superuser mode and it seemed to install fine. Then tried tom's 2 commands, and then ./circuitjs1, which still gave the same error regarding libX11.xcb1.

(base) kg@kg-Latitude-E7450:~$ npm install electron -g
loadDep:global-tunnel-ng  ▌ ╢██████████░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░╟
WARN engine global-agent@2.1.8: wanted: {"node":">=10.0"} (current: {"node":"8.1loadDep:serialize-error → ▐ ╢███████████░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░╟
WARN engine semver@7.1.3: wanted: {"node":">=10"} (current: {"node":"8.10.0","npnpm WARN checkPermissions Missing write access to /usr/local/lib
/usr/local/lib
└─┬ electron@8.1.1 
  ├─┬ @electron/get@1.9.0 
  │ ├─┬ debug@4.1.1 
  │ │ └── ms@2.1.2 
  │ ├── env-paths@2.2.0 
  │ ├─┬ fs-extra@8.1.0 
  │ │ ├── graceful-fs@4.2.3 
  │ │ ├── jsonfile@4.0.0 
  │ │ └── universalify@0.1.2 
  │ ├─┬ global-agent@2.1.8 
  │ │ ├── boolean@3.0.1 
  │ │ ├── core-js@3.6.4 
  │ │ ├── es6-error@4.1.1 
  │ │ ├─┬ matcher@2.1.0 
  │ │ │ └── escape-string-regexp@2.0.0 
  │ │ ├─┬ roarr@2.15.2 
  │ │ │ ├── detect-node@2.0.4 
  │ │ │ ├─┬ globalthis@1.0.1 
  │ │ │ │ └─┬ define-properties@1.1.3 
  │ │ │ │   └── object-keys@1.1.1 
  │ │ │ ├── json-stringify-safe@5.0.1 
  │ │ │ ├── semver-compare@1.0.0 
  │ │ │ └── sprintf-js@1.1.2 
  │ │ ├── semver@7.1.3 
  │ │ └─┬ serialize-error@5.0.0 
  │ │   └── type-fest@0.8.1 
  │ ├─┬ global-tunnel-ng@2.7.1 
  │ │ ├── encodeurl@1.0.2 
  │ │ ├── lodash@4.17.15 
  │ │ ├─┬ npm-conf@1.1.3 
  │ │ │ ├─┬ config-chain@1.1.12 
  │ │ │ │ ├── ini@1.3.5 
  │ │ │ │ └── proto-list@1.2.4 
  │ │ │ └── pify@3.0.0 
  │ │ └── tunnel@0.0.6 
  │ ├─┬ got@9.6.0 
  │ │ ├── @sindresorhus/is@0.14.0 
  │ │ ├─┬ @szmarczak/http-timer@1.1.2 
  │ │ │ └── defer-to-connect@1.1.3 
  │ │ ├─┬ cacheable-request@6.1.0 
  │ │ │ ├── clone-response@1.0.2 
  │ │ │ ├── get-stream@5.1.0 
  │ │ │ ├── http-cache-semantics@4.1.0 
  │ │ │ ├─┬ keyv@3.1.0 
  │ │ │ │ └── json-buffer@3.0.0 
  │ │ │ ├── lowercase-keys@2.0.0 
  │ │ │ ├── normalize-url@4.5.0 
  │ │ │ └── responselike@1.0.2 
  │ │ ├── decompress-response@3.3.0 
  │ │ ├── duplexer3@0.1.4 
  │ │ ├─┬ get-stream@4.1.0 
  │ │ │ └─┬ pump@3.0.0 
  │ │ │   ├── end-of-stream@1.4.4 
  │ │ │   └─┬ once@1.4.0 
  │ │ │     └── wrappy@1.0.2 
  │ │ ├── lowercase-keys@1.0.1 
  │ │ ├── mimic-response@1.0.1 
  │ │ ├── p-cancelable@1.1.0 
  │ │ ├── to-readable-stream@1.0.0 
  │ │ └─┬ url-parse-lax@3.0.0 
  │ │   └── prepend-http@2.0.0 
  │ ├── progress@2.0.3 
  │ ├─┬ sanitize-filename@1.6.3 
[... 128 lines cut ...]
  │ ├── progress@2.0.3 
  │ ├─┬ sanitize-filename@1.6.3 
  │ │ └─┬ truncate-utf8-bytes@1.0.2 
  │ │   └── utf8-byte-length@1.0.4 
  │ └── sumchecker@3.0.1 
  ├── @types/node@12.12.30 
  └─┬ extract-zip@1.6.7 
    ├─┬ concat-stream@1.6.2 
    │ ├── buffer-from@1.1.1 
    │ ├── inherits@2.0.4 
    │ ├─┬ readable-stream@2.3.7 
    │ │ ├── core-util-is@1.0.2 
    │ │ ├── isarray@1.0.0 
    │ │ ├── process-nextick-args@2.0.1 
    │ │ ├── safe-buffer@5.1.2 
    │ │ ├── string_decoder@1.1.1 
    │ │ └── util-deprecate@1.0.2 
    │ └── typedarray@0.0.6 
    ├─┬ debug@2.6.9 
    │ └── ms@2.0.0 
    ├─┬ mkdirp@0.5.1 
    │ └── minimist@0.0.8 
    └─┬ yauzl@2.4.1 
      └─┬ fd-slicer@1.0.1 
        └── pend@1.2.0

pfalstad, 2020-03-15:
Go to the directory where circuitjs1 is and run electron.

gskrish2001, 2020-03-15:
So I had some problems installing electron (i.e. I had installed inside .wine subfolders for some reason). But then I went through (followed this video: https://www.youtube.com/watch?v=qsUH3j6xe_Y) and installed it under a non-virtual drive (such as under wine)
 and now I am able to open it. So basically running:
(base) kg@kg-Latitude-E7450:~/Downloads/circuitjs1$ electron

Electron 8.1.1 - Build cross platform desktop apps with JavaScript, HTML, and CSS
Usage: electron [options] [path]

A path to an Electron app may be specified. It must be one of the following:
  - index.js file.
  - Folder containing a package.json file.
  - Folder containing an index.js file.
  - .html/.htm file.
  - http://, https://, or file:// URL.

Options:
  -i, --interactive     Open a REPL to the main process.
  -r, --require         Module to preload (option can be repeated).
  -v, --version         Print the version.
  -a, --abi             Print the Node ABI version.

[image]

So I also tried opening circuitjs1 using the command given therein (in the picture):
(base) kg@kg-Latitude-E7450:~/Downloads/circuitjs1$ /usr/local/lib/node_modules/electron/dist/electron /home/kg/Downloads/circuitjs1/circuitjs1
App threw an error during load
/home/kg/Downloads/circuitjs1/circuitjs1:1
ELF
^

SyntaxError: Invalid or unexpected token
    at Module._compile (internal/modules/cjs/loader.js:896:18)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:986:10)
    at Module.load (internal/modules/cjs/loader.js:816:32)
    at Module._load (internal/modules/cjs/loader.js:728:14)
    at Module._load (electron/js2c/asar.js:717:26)
    at Function.Module._load (electron/js2c/asar.js:717:26)
    at loadApplicationPackage (/usr/local/lib/node_modules/electron/dist/resources/default_app.asar/main.js:109:16)
    at Object.<anonymous> (/usr/local/lib/node_modules/electron/dist/resources/default_app.asar/main.js:155:9)
    at Module._compile (internal/modules/cjs/loader.js:968:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:986:10)
Gtk-Message: 15:02:00.428: GtkDialog mapped without a transient parent. This is discouraged.

[image]

The app seems to be running since trying to close the above window says the app is running on the commandline. What should I do? Thank you.

lkz0ne, 2020-04-26:
Hi falstad. I love the app.
I'm in linux version I am having segmentation fault.

`root@kali:~/Downloads/circuitjs1# ./circuitjs1 --no-sandbox
Segmentation fault`

lkz0ne, 2020-04-26:
Hi falstad. I love the app.
I'm in linux version and I am having segmentation fault.

'root@pc:~/Downloads/circuitjs1# ./circuitjs1 --no-sandbox Segmentation fault'

pfalstad, 2020-04-26:
I don’t have Linux so I can’t help much with that, sorry.

lkz0ne, 2020-04-26:
`node /usr/local/lib/node_modules/electron/install.js` solved it. But now i am having same as gskrish2001

pfalstad, 2020-04-26:
I should have looked at his message more carefully.  Try:

% electron circuitjs1/resources/app/index.html

lkz0ne, 2020-04-26:
Thanks for the reply, pfalstad.
Everything went well. No execution errors. But it came on an all white screen.