sharpie7/circuitjs1 #394: User Manual

SM-Fahim, 2019-11-17:
I badly need a user manual as a new user of this software. I'm trying to do 2 tasks but stuck.
1. Using voltage control voltage source: Don't understand how to connect.
2. Need to show 2 different values of voltage source and the corresponding current in a single graph to compare them.

pfalstad, 2019-11-18:
To view I1 vs V1 you can just right click on the voltage source and select View in Scope, right click on the scope and select Properties, then View V vs I.  It won't be much help here though because it doesn't show a scale.

To view I2 vs V1 you would need to plot I2 and V1 seperately, then combine them (right click on the second one and select Combine), then Properties->plot XY.  Again, there's no scale shown and you can't match it to the first graph so it won't be much help.

SM-Fahim, 2019-11-18:
Okay sir, I solved my problem separately. i'll be deleting this issue, but before that, could you tell me how can I use an AC current source? I need it to represent 5cos2t A. Couldn't find it.

pfalstad, 2019-11-18:
here you go: http://tinyurl.com/vsf79ee

create voltage-controlled current source, set function to 5*cos(2*t), no inputs.

SM-Fahim, 2019-11-18:
Actually I'm trying to solve this circuit from last night, without any rest, and didn't see any improvement. Can you help me? And I can't set the value for that current source. It would be very helpful for me if you help me.
[image]

sharpie7, 2019-11-21:
There are tutorial videos for basic operations in my page: http://lushprojects.com/circuitjs/

closed 2019-11-23 by sharpie7