sharpie7/circuitjs1 #396: cpu usage

ttt31, 2019-11-20:
CPU usage rate is only 19%, can you let this process faster? I want to use up this extra power.
Environment: Inspiron 15 3000 (AMD® Ryzen)

pfalstad, 2019-11-21:
And you have the speed turned all the way up?  Must be a simple circuit.

colinhowell, 2019-11-21:
My guess would be that this person's CPU usage is counting activity of all their CPU cores; the simulator can only use one of them.

ttt31, 2019-11-21:
Simulates a fairly elaborate circuit.  It's a shame that only 1 core can be used

sharpie7, 2019-11-21:
Ultimately it is the browser that provides the run-time environment for the Javascript. AFAIK all the current generation of browsers are limited to one core per tab. If that was to change then it might be useful to look at our code and see if we could do anything to help parallelize the calculations, but at the moment I don't know of anything we could do in the code that would improve the situation.

pfalstad, 2019-11-21:
I think the algorithm would be hard to parallelize even if we had access to multiple cores.

Depending on the circuit, you may be able to increase the time step size to speed things up.  (in Options->Other Options...)  Increasing the time step 2x will make it run twice as fast, but would be less accurate.  For a digital circuit it usually doesn't matter.  For an analog circuit, it depends, but I'd say the timestep should be around 1/(32*f) where f is the max frequency generated by any part of the circuit.

ttt31, 2019-11-22:
This circuit simulator is very easy to use.  The download version seems to be able to handle it, but it seems that the program will be rewritten, so the update effort seems to be amazing🤔

closed 2019-11-23 by sharpie7