sharpie7/circuitjs1 #400: Subcircuit

jmek94, 2019-12-02:
A few months ago I began using the subcircuit tool in the simulator. I used it to simulate flip-flops and logic gates for a discrete component counter circuit I was building. I haven't had to use the subcircuit simulator for a while, but recently found myself needing to use it again.

As I always have I would create the circuit, use labelled nodes for I/O, highlight the components and be prompted with an editable model. Now after finishing and selecting 'ok', the 'black box' subcircuit would _not_ appear, as it had, and my schematic remained unchanged. I tried it with several different types of circuits with different components. I haven't even had luck with the FBR circuit example linked in the sub-circuits page.

https://www.falstad.com/circuit/subcircuits.html

Were there any updates that may have affected this feature? Could something else be affecting this? Am I the only one with this issue?

Thanks

pfalstad, 2019-12-02:
It's working for me.  There haven't been any updates that might affect this since July 9.

So when you go to the sub-circuits example page, and click on "here is an example using the rectifier", what happens?

The black box doesn't appear unless you select Active Building Blocks->Subcircuit Instance from the right-mouse menu, but that wouldn't explain why the example circuits don't work for you.

jmek94, 2019-12-02:
It's working for me now. For some reason I was incorrectly remembering the subcircuit being immediately replaced by the black box. Using Active Building Blocks ->Subcircuit Instance works as it should. 

I was a little confused because the [subcircuits page](https://www.falstad.com/circuit/subcircuits.html)   doesn't mention  Active Building Blocks->Subcircuit Instance, only  File->Create Subcircuit

Thanks for the quick response.

closed 2019-12-02 by jmek94

pfalstad, 2019-12-03:
Yes, I'll add some more explanation to that page.

balazsmolnar72, 2022-07-15:
Could you please document in the [subcircuits page](https://www.falstad.com/circuit/subcircuits.html) 

1. When you create subcircuit and check the Save Across Sessions box, where the subcircuit will be stored in file system?
2. If I would like to take that out from the list of subcircuits how can I do this?
3. For the pins do I need to use labeled node or something else? (it is not clear at the documentation)
4. Can I recall somehow the subcircuit if I would like to edit it (for enhacement)?

pfalstad, 2022-07-15:
1. in your browser local storage.  Using chrome, right click on the side bar and pick "Inspect", then from the developer panel that opens up, click on "Application" (you may have to click on >> to see it).  Then select local storage, http://www.falstad.com, and look for keys of the form subcircuit:name 
2. find the key, right click it and select Delete
3. labeled node
4. yes, if you check the "Save Across Sessions" checkbox.  create a subcircuit instance, right click it, Edit, Load Model Circuit

balazsmolnar72, 2022-07-15:
Thank you very much for your quick response ! 
I'm using the installed desktop version 1.2.2 not the browser version in this case where are the subcircuits are stored?
All my other questions are answered thank you!

pfalstad, 2022-07-15:
Options->Toggle Dev Tools
click on "Application" (you may have to click on >> to see it)
select local storage, file://, look for keys of form subcircuit:name

Bagunceiro, 2024-01-10:
When I click Options->Toggle Dev Tools nothing happens, so I can't select "Application" (or click >>). What am I doing wrong?

Version is  2.7.2js, running under Linux Mint (5.15.0-91-generic #101-Ubuntu SMP Tue Nov 14 13:30:08 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux).

Thanks

pfalstad, 2024-01-10:
Apparently this is an old bug in the linux version of electron.  It has since been fixed.

The circuitjs1 binary in the offline version is just electron.  You can do npm install electron@v28.1.2 to install the latest version of electron and then run electron from inside the circuitjs1 directory.  Maybe that will work.  (I don't have linux so I can't try it out)

AgainPsychoX, 2024-03-26:
> 1. in your browser local storage.  Using chrome, right click on the side bar and pick "Inspect", then from the developer panel that opens up, click on "Application" (you may have to click on >> to [...]

Is it possible to add this information to the https://www.falstad.com/circuit/subcircuits.html document? 

Took me 10 minutes of Googling to find some information included.