sharpie7/circuitjs1 #401: Bug: No 2bits digital counter

programistawpf, 2019-12-04:
I am trying change number of bits digital counter to 2 . Defaultis 4 . Edit>> #of Bits 2 >>>No changes. Is posible change to 3 bits or 5 ....
Thanks Paul.

pfalstad, 2019-12-04:
3 is the minimum number of bits.  If I allow you to set it to 2, then that would make it slightly more complicated to lay out the pins.  You can set it to 3 and just ignore the extra bit.

programistawpf, 2019-12-05:
W dniu 2019-12-04 o 21:38, Paul Falstad pisze:
> 3 is the minimum number of bits. If I allow you to set it to 2, then 
> that would make it slightly more complicated to lay out the pins. You 
> can set it to 3 and just ignore the extra bit.

It is true , Thanks :)

closed 2019-12-06 by sharpie7