sharpie7/circuitjs1 #403: Can the LED resistance be more realistic?

Galaxy-Prism, 2019-12-06:
The LED Resistance is absurdly small. 

[image]

pfalstad, 2019-12-06:
Generally, the devices in this simulator are ideal, so if you want some resistance you have add it explicitly.  In the case of LEDs and diodes though, you can have them use more realistic models with resistance.  You just have to create a custom model for it.

Do you have a particular LED device in mind?

Galaxy-Prism, 2019-12-06:
No. I just thought that less than 1 ohm was unrealistic for an LED

sharpie7, 2019-12-06:
LEDs are not resistors and don't have a resistance. You might find this video helpful to understand the behaviour of LEDs and other diodes: https://www.youtube.com/watch?v=Pp5AAj2aIzc

If you connect an LED directly to a low impedance voltage source higher than its threshold voltage it wll pass a very large current (normally for a fraction of a second before being destoryed).

closed 2019-12-06 by sharpie7