sharpie7/circuitjs1 #404: Feature- Switching regulator sample circuit

Circuitsoft, 2019-12-08:
For anyone who would like to simulate the activity of a buck regulator, this seems to be a good start. It may be worth adding to the "Circuits" menu.

https://www.falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+1.700203994009402+37+5+50%0Ad+464+336+496+304+2+default%0Ad+496+368+464+336+2+default%0Ad+496+368+528+336+2+default%0Ad+528+336+496+304+2+default%0Aw+528+336+544+336+0%0Aw+544+336+544+288+0%0Aw+544+288+384+288+0%0Av+384+288+384+384+0+1+60+19.8+0+0+0.5%0Aw+384+384+448+384+0%0Aw+448+384+448+336+0%0Aw+448+336+464+336+0%0Ag+496+368+496+416+0%0Aw+496+304+496+272+0%0Aw+496+272+544+272+0%0Al+544+272+624+272+0+0.000022+2.3629428453021784%0Aw+624+272+640+272+0%0Ac+640+272+640+400+0+0.00015999999999999999+11.046044497938206%0Ag+640+400+640+416+0%0Aw+640+272+704+272+0%0Af+752+336+752+272+41+1.5+20%0Aw+768+272+800+272+0%0Al+800+272+960+272+0+0.0047+1.051680941603237%0Ad+800+320+800+272+2+default%0Ag+800+320+800+336+0%0Ac+960+272+960+320+0+0.00009999999999999999+3.3670477641537557%0Ag+960+320+960+336+0%0Aw+960+272+1072+272+0%0Ar+1072+272+1072+320+0+3.3%0Ag+1072+320+1072+336+0%0Aw+960+272+896+368+0%0AR+896+400+976+400+0+4+4000+0.05+3.3+0+0.5%0Aw+704+272+736+272+0%0Ar+704+272+704+336+0+100000%0Aw+704+336+752+336+0%0Aw+752+368+752+336+0%0Af+800+384+752+384+32+1.5+20%0Ag+752+400+752+416+0%0Aa+896+384+800+384+8+5+0+1000000+3.3670477641537557+3.2800000007448222+100000%0Ao+27+1+7+28930+0.0001+0.0001+0+1+3.7304511992352776%0Ao+16+1+0+4355+20+6.4+0+2+16+3%0Ao+30+1+0+12546+3.3480000008595674+0.0001+0+2+30+3%0A

sharpie7, 2020-01-02:
Thanks - I always thought it would be good to have the standard regulator architectures in the examples.

When you say this is a "good start", are there specific things you are thinking still need work?

Circuitsoft, 2020-01-02:
It seems to work-ish, but it's not the most stable thing out there. I kinda hacked it together quickly. It has large voltage swings on load changes, because I haven't tuned the time domain of the feedback loop. Trying to make it more responsive (mostly with higher switching frequency) does seem to run into processing time issues.

ZVSphiler, 2022-01-09:
Oh yea I made a similar thing an isolated SEPIC converter with a simple current mode control loop but I'm not sure about stability and correctness . I just bodged it together. But for the case it is worth being put into the circuits menu feel free to do it.

https://tinyurl.com/y5rrw4u9