sharpie7/circuitjs1 #408: Bug - CMOS Inverter Subcircuit does not work

giulioscattolin, 2020-01-10:
Given a subcircuit of [CMOS_Inverter.txt](https://github.com/sharpie7/circuitjs1/files/4045357/CMOS_Inverter.txt), this is what happen when it is used:

[image]

It makes no sense to me since `in`, `out` and `vee` are at `4.987 V` while they shouldn't be connected to anything, yet.  Moreover, `vee` it draws a very high current (~ GA) if grounded..

Is this a bug or am I missing something? Thank you very much!

sharpie7, 2020-01-25:
Yes, I see the same problem with your example. Odd! @pfalstad is the expert on subcircuits

pfalstad, 2020-01-25:
Sorry, this email got lost in my inbox.  Something definitely seems wrong here.  I'll take a look.

pfalstad, 2020-01-25:
Yeah this is a bug in the MOSFET body diode code.  The body terminal wasn't getting set when using a MOSFET in a subcircuit, so it was using the gate as the body terminal  Uncheck "simulate body diode" to work around this for now, I'll fix.

closed 2020-02-09 by sharpie7