sharpie7/circuitjs1 #412: How to run on eclipse

raquelhortab, 2020-01-27:
Hi, I have opened the project and added the GWT plugin. However, I am missing on how to run the project on development. In the README it is said that:

> Running in super development mode is done by clicking on the "run" icon on the toolbar and choosing http://127.0.0.1:8888/circuitjs.html from the "Development Mode" tab which appears.

But this is just not how eclipse behaves, maybe it is for older versions? I cannot run if I don't have a run configuration. Could someone explain how to run it? Which kind of run configuration do I need to set up?

Thanks in advance

pfalstad, 2020-01-27:
Try clicking on circuitjs1 in the package explorer on the left (the top level root) and then right click, then select "Run As ....", then "GWT Development Mode with Jetty".  Then it should ask you to select an item to open, so you pick circuitjs.html.  Does that work?  If not, what happens if you try it?

NulAsh, 2020-02-08:
> Could not find the main class. Program will exit.

Also I see the following problems:
```
Description	Resource	Path	Location	Type
The method copyOf(double[], int) is undefined for the type Arrays	CirSim.java	/circuitjs1/src/com/lushprojects/circuitjs1/client	line 5218	Java Problem
The method isEmpty() is undefined for the type String	CirSim.java	/circuitjs1/src/com/lushprojects/circuitjs1/client	line 2956	Java Problem
The method isEmpty() is undefined for the type String	CirSim.java	/circuitjs1/src/com/lushprojects/circuitjs1/client	line 4246	Java Problem
The method isEmpty() is undefined for the type String	CustomCompositeElm.java	/circuitjs1/src/com/lushprojects/circuitjs1/client	line 52	Java Problem
The method isEmpty() is undefined for the type String	CustomCompositeElm.java	/circuitjs1/src/com/lushprojects/circuitjs1/client	line 53	Java Problem
The method isEmpty() is undefined for the type String	CustomCompositeElm.java	/circuitjs1/src/com/lushprojects/circuitjs1/client	line 62	Java Problem
The web.xml file does not exist	WEB-INF	/circuitjs1/war	Unknown	GWT Web App Problem
```

pfalstad, 2020-02-08:
Yeah the web.xml got left out of the repository, probably because that whole directory is listed in .gitignore.  I just added it to my fork.

I'm not sure what's going on with the other errors.  What version of java do you have?

NulAsh, 2020-02-08:
I have a lot of versions, but it want to use 1.5