sharpie7/circuitjs1 #413: transistor Saturation & fwd active regions reversed

2shaar2059, 2020-01-30:
When I use an NPN BJT transistor with Vbe = 5V and Rb = 1000 ohms, Vce = 20V and Rc = 1 ohm, hovering my mouse over the transistor shows that the transistor is in the fwd active region when it should actually be in the saturation region.

When I use an NPN BJT transistor with Vbe = 5V and Rb = 1000 ohms, Vce = 20V and Rc = 1000 ohm, hovering my mouse over the transistor shows that the transistor is in the saturation region when it should actually be in the fwd active region.

pfalstad, 2020-01-30:
The definitions of transistor regions we're using are roughly the same as these: https://en.wikipedia.org/wiki/Bipolar_junction_transistor#Regions_of_operation

What definition are you using for saturation?

2shaar2059, 2020-01-30:
Thanks you so much for the speedy reply. I love your site!!!

This is the defintion im using:

https://www.electronics-tutorials.ws/transistor/tran_4.html

From what i understand, if the collector current is at its maximum (base
current * beta) the transistor is said to be saturated.

pfalstad, 2020-01-30:
If you scroll down it says "saturation characteristics: max collector current flows (Ic = Vcc/RL)".  But Vcc/RL in this case is 20 amps.  In the first case, with Rc = 1 ohm, Ic is much less than that.

It says Vce = 0 in ideal saturation.  But in the first case with Rc = 1 ohm, Vce is almost 20 volts.

The second case you mention (Rc = 1k) has a much smaller Vce, and Ic is much closer to Vcc/RL.

closed 2020-02-09 by sharpie7