sharpie7/circuitjs1 #419: GWT module 'circuitjs1' may need to be (re)compiled

taffarel55, 2020-02-19:
Anybody?

pfalstad, 2020-02-19:
What?  Are you seeing this message somewhere?  More details please!

taffarel55, 2020-02-19:
I'm seeing this message after initial compilation. That's a js alert

[image]

pfalstad, 2020-02-19:
Not sure, haven't seen that.  How are you running it?  To run it, I right click on the root node, "circuitjs1" in the package explorer on the left, then select "Run As" and "GWT Development Mode With Jetty".  Then when it asks what to open, I select circuitjs.html.  Then double-click to open the URL it displays.  It says "compiling" and then shortly after that, the applet starts.  Is any of that happening?

You could also try following the advice here: https://stackoverflow.com/questions/5719118/gwt-module-may-need-to-be-recompiled-redux

closed 2020-02-21 by taffarel55