sharpie7/circuitjs1 #424: Faulty relay operation

quickscanner, 2020-03-25:
New to this app. I am an automotive trainer. Recently started using the app for electrical training, in the place of the conventional simulator board. It is amazing, just like from a dream. I have a problem with a circuit I made recently. It is a faulty central door lock system circuit meant for troubleshooting practice. There are two faults set . A fuse [fuse 2]that blows when ignition is ON and FL unlock function not operating when fault 1 is rectified. I had to use LEDs to represent lock and unlock operation as I could not find a DC motor anywhere. My problem is that the lock and unlock relays are turning ON when the ignition switch is turned ON with the fault set [it is a short to ground for fuse 2]. My electrical background is very poor. May be I am missing something fundamental. I am not expecting any current in the relay control circuit with just the ignition switch turned ON. How they are turning ON? Please find below the circuits in text.
[Circuit - central door lock.txt](https://github.com/sharpie7/circuitjs1/files/4379329/Circuit.-.central.door.lock.txt)
[Circuit - with faults.txt](https://github.com/sharpie7/circuitjs1/files/4379330/Circuit.-.with.faults.txt)

pfalstad, 2020-03-26:
I'm not sure I'm understanding the problem.  Could you provide more detail?

I tried loading "Circuit - with faults" and closed the ignition switch.  I see a short directly to ground through fuse 2.  That's expected, right?

Because of all the current through fuse 1, the input to the two NAND gates is too low, and is treated as a low input.  (The threshold is half the output voltage, 2.5 V in this case.)  So the NAND outputs are high, which causes the relays to be turned on.  That's why you're seeing current through the relay control circuit.

quickscanner, 2020-03-26:
Yes, short to ground through fuse 2 is intentional.
I was not aware of the threshold voltage of NAND gates. I was a a little surprised, this problem was not there when I changed the cell voltage from 5 to 12. Now I understand why!
Thank you.

closed 2020-03-26 by quickscanner