sharpie7/circuitjs1 #425: LM324 model

Felthry, 2020-03-25:
The "op amp (real)" element appears to function as a subcircuit. The model of the LM741 is accessible from the examples (at least I assume it's the same model), but the model of the LM324 doesn't appear to be. Is there any way we could see what precisely is going on inside the model?

Beyond just my own curiosity, it'd be didactically useful to be able to show the differences between the internals of a really bad op amp (the 741) and a less bad one (the 324).

pfalstad, 2020-03-25:
Yes I should put it in the examples.  Here it is.

[circuit-324.txt](https://github.com/sharpie7/circuitjs1/files/4383821/circuit-324.txt)

Felthry, 2020-03-26:
While looking at that model, I figured out what seems to be the cause of the simulation running poorly when the 324 model is selected and some pins are left floating on it. If you put a large-value resistor (10M or so will do) in parallel with the leftmost, 6 μA current source to limit its output impedance, the simulation no longer gets runaway voltages with the model unpowered. It still gives an unphysical 60 volt differential between the supply rails and the inputs, but it no longer makes the entire simulator slow to a crawl or fail to converge.

I'm assuming you're drawing current sources instead of actual current mirrors like would be used in the real circuit to minimize model complexity?

pfalstad, 2020-03-26:
I think the schematic I was using had current source symbols rather than the actual implementation.  I see the problem though.. It's not a "real" op amp with those elements in there.

Felthry, 2020-03-27:
Would you like me to modify the model to use current mirrors? I should be able to do that fairly easily--and what with current events, I have an unusual amount of free time on my hands.

Felthry, 2020-03-27:
In fact, I went ahead and did that. I couldn't find any information on exactly what type of current source is used in these op amps, so I went with one that I think wouldn't be too difficult to make in a bipolar process: the zener-diode based current source (however, to maintain the 324's rated voltage range, I had to use a 1.8 V zener diode! low-voltage zener diodes have very soft knees but this one should be okay as the current through it is pretty much constant). I've attached the circuit I ended up with, along with a demonstration of it working as a unity-gain buffer (demonstrating the slew rate, which is approximately equal to what TI's datasheet says it should be), and the circuit I used for testing the voltage stability of the current sources I used.

[real-324.txt](https://github.com/sharpie7/circuitjs1/files/4394167/real-324.txt)
[real-324-test.txt](https://github.com/sharpie7/circuitjs1/files/4394168/real-324-test.txt)
[testing current source.txt](https://github.com/sharpie7/circuitjs1/files/4394169/testing.current.source.txt)

pfalstad, 2020-03-27:
That's great, thanks!  I'll take a look at this...

pfalstad, 2020-12-14:
Ok I am finally looking at this.  It works great, but I need to be able to adjust the slew rate and current limiting.  Adjusting the slew rate is easy by changing the capacitor.  But this model's current limiting is not so good. Is that how a real LM324 behaves?

https://tinyurl.com/y7fvvxlf

Felthry, 2020-12-22:
I'm not actually sure, and don't have a 324 to hand to check, sorry. This looks like (a particularly severe case of) phase inversion, though, which is a very common problem on old op amps (modern ones don't have it), so I wouldn't be surprised if that is how a real one behaves.

closed 2022-07-11 by pfalstad