sharpie7/circuitjs1 #427: Export V/I

mrquincle, 2020-03-26:
Hi, we're working on load identification using V/I trajectories. It's cool how you can visualize all the different ones. For example, consider this set of loads (it's from a paper, I'll be able to get it for you if you're interested in it).

[image]

It is cool to try to create the different plots. The fun ones to try to create are the higher-order ones. Something like (e) but not exactly:

[image]

We're subsequently running classifiers on those VI trajectories. What would we need to do to extract the data from the (falstad) simulator, to do so? Can we obtain it through the browser's console somehow? 

Or even better. Is it a nice idea to have export functions for current/voltage data in the menu?

mrquincle, 2020-03-26:
Ah, I found out I can export voltages via "Draw", "Outputs and Labels", "Add Data Export". I then can export voltage information, http://tinyurl.com/wawqn4v.

Can I export current data as well? Preferably synced. :-)

pfalstad, 2020-03-26:
You just need to use a current controlled voltage source to convert the current to voltage:

http://tinyurl.com/u83gq83

mrquincle, 2020-03-26:
Sweet!! That works splendidly! Let's see if I can bump the resolution on this thing.

[image]

mrquincle, 2020-03-26:
Mmm... It is not possible to sync it somehow I guess?

pfalstad, 2020-03-26:
Options->other options will let you set the time step size to a smaller value.

pfalstad, 2020-03-26:
If you click reset first, then both data outputs will start from the same point...

mrquincle, 2020-03-26:
It is fabulous!! Really great. http://tinyurl.com/vgfgs5f I set it at 1us and enlarged the buffer to around 100.000 data points. This is the result.

[image]

Thank you so much!

mrquincle, 2020-03-26:
Starting with black dots, to show time progression...

[image]

Feel free to close this issue!

mrquincle, 2020-03-26:
If you want to play with this yourself. Here's an R script: https://gist.github.com/mrquincle/e00ebd6034a4c565446c07480ab28af6

mrquincle, 2020-03-27:
A very low-priority request would then be to perhaps allow also for an easier way to export voltage/current simultaneously (without adjusting the circuit, clicking restart, and right clicking different export labels, and hoping that you've synced data). I often downloaded either the same data, or not synced data, or made other dumb mistakes. :-)

closed 2020-04-03 by mrquincle