sharpie7/circuitjs1 #428: Polarized Capacitor?

madcountess, 2020-03-26:
I'm trying to work on some analog circuits I am yet to be confident of, and a few of the circuits I've found that I want to test have polarized capacitors, like electrolytic capacitors, and I'm wondering how I can either polarize a nonpolar cap or something. I've searched through the directions page and I'm just kinda lost.

madcountess, 2020-03-26:
Nevermind. I am using an apparently incomplete build. I'll have to take it up with the maintainer of the arch/manjaro package

closed 2020-03-26 by madcountess

pfalstad, 2020-03-26:
Polarized capacitors are the same as regular capacitors, except that they stop the simulation and warn you if you have a negative voltage across them.  So you don't need them unless you're designing a circuit you want to build out of real components.