sharpie7/circuitjs1 #431: feature request/component: Power meter

Felthry, 2020-03-29:
While there are ammeters, voltmeters, and even ohmmeters that can be placed in the simulation, I feel like a wattmeter would be a useful addition. It's easy to see the power consumption of a single element, yes, but it would be nice to have a component that can tell you the total power passing through it (bonus points if there's also a three-phase power meter!). I've made a mockup of what I'm thinking of below; the load in either case can be any combination of components, so you don't have to sum a bunch of numbers if you want to know the power going past a certain point in the circuit.

Being able to make it also measure reactive power would be nice, but I'm not sure how one would do that since reactive power is only defined as an average, not instantaneous value.

[image]

An example of how this would let you do something you can't do by just looking at the power consumption of the source:

[image]

Wattmeters here would allow you to examine the power loss in the line.

Now, as an alternative to this entirely that might be both easier to implement and more useful overall: consider making the ability to have a scope show the total power consumption of a number of selected elements. Select several components in a group, then "add total power monitor" could replace "view in scope", since "view in scope" already doesn't work if you have multiple things selected.

...in fact, the more I think about this, the more I think it's a better option than adding a power meter as an actual element you put into the circuit. The fact that you can only select multiple things in a box selection rather than a multi-select with clicks might be the only sticking point.

pfalstad, 2020-03-29:
Or a sum checkbox in the scope properties that sums whatever plots are present in the scope.

Felthry, 2020-03-29:
A sum option would probably work quite well, yes.