sharpie7/circuitjs1 #432: Feature request/scope: RMS and average as separate options, plus other calculations

Felthry, 2020-03-30:
There are cases when it would be helpful to be able to see the average value of a voltage or current, rather than just the RMS value. Consider a PWM signal, for instance, or simply a noisy signal that you want to average out the noise on.

As the scope works now, you can display the average power, or the RMS voltage or current, but you cannot display the average voltage or current (you also can't display the RMS power, but I don't think RMS power is a meaningful quantity in any circumstances).

Perhaps as a bonus, being able to show the peak-to-peak voltage would be nice as well. You can have the scope display both peaks, and then manually subtract to find that, but it'd be convenient to have the scope just tell you. Similarly, it'd be nice to be able to AC-couple the scope's RMS calculation, so you can see that this signal has so-and-so DC offset using the average voltage function, and it has however-much RMS AC on top of it.

I know I'm submitting a lot of feature requests lately--I hope that isn't a problem! Current events mean I have a lot of free time on my hands, and some of that is spent messing around with circuitry stuff--meaning that I find myself thinking "this would be a nice feature to have" more often than normal.

mike0209-eng, 2020-11-03:
I am looking for very similar features which would make simulation much easier: A voltmeter, which can be switched to RMS and to average-value too. 
A question here: why is the RMS value changing sometimes, even when the circuit is in steady-state and the values are not supposed to change? Look at the voltage drop at the 40 Ohm load in the rectifier circuit attached for an example.
[circuit-20201103-1540(rms show).circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/5481907/circuit-20201103-1540.rms.show.circuitjs.txt)

pfalstad, 2020-11-03:
Apparently the voltmeter doesn't do a good job of RMS averaging if the waveform is not a sine wave.  If you view it in scope and select "show RMS average" you get a more steady value.  I'll have to copy that code over if possible.

mike0209-eng, 2020-11-04:
[test_rms_rect_w_offset.txt](https://github.com/sharpie7/circuitjs1/files/5486366/test_rms_rect_w_offset.txt)
I just had a look on the RMS values in my simple circuit and noted the following (please use the attached file):
- set the voltage to AC (amplitude 10 V) with an offset of 10 Vdc: Voltage RMS and current RMS are correct (and including the offset correctly).
- set the voltage to "Square Wave", keeping the offset. Now the RMS value change between two values. Apparently the non-zero value is also strange (447 mV rms on a 20 Ohms resistance, while the voltage is 20 volts, same for the amperemeter's value).
This behaviour of cycling between two values continues if you reduce the DC offset to 5 V (i.e. voltage goes below zero), but in this case the displayed RMS-values make more sense.
Can you please check whether you can reproduce this? I use 2.2.18.js in the standalone-version.

pfalstad, 2020-11-04:
yes, I'm seeing that too.

mike0209-eng, 2020-11-05:
Good, so it's at least reproducible.