sharpie7/circuitjs1 #434: Component Request

Jaguem, 2020-04-07:
I didn't find voltage regulators like 7805,7812,7905,7912,LM317. Could you add those components?

JakoI3, 2020-06-03:
I can only agree, I would also like to have the components in it.

Daktin, 2020-06-05:
Here is all the parts from representative schematic diagram in lm317 datasheet, it kinda work, but needs tweaking:

[lm317.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/4733954/lm317.circuitjs.txt)

pfalstad, 2020-09-16:
Thanks Datkin.  Here is that schematic as a subcircuit:

http://tinyurl.com/yytwm2hy
http://tinyurl.com/y3ndq8n8

pfalstad, 2020-09-16:
There's a 7805 model here if anyone wants to make a subcircuit out of it: https://www.youspice.com/spiceprojects/spice-simulation-projects/power-and-battery-circuits-spice-simulation-projects/regulated-power-supplies-spice-simulation-projects/spice-model-of-lm78xx-voltage-regulators/