sharpie7/circuitjs1 #435: Three-phase transformer and problem with transfiguration from delta to star and star to delta.

doomfan, 2020-04-15:
Hello. Off course very nice simulator for electronic circuit, but I saw one problem - when I try to use three transformers and combine them into one whole like for one three-phase transformer I saw problem. What is this problem ? For example, when I connect primary transformers windings in delta circuit and I connect secondary transformers windings in star circuit and finally I connect voltmeters to secondary transformers windings voltages are same like on primary transformers windings. Transfiguation so not work properly like for me. For example if on primary transformers windings voltages are about 400 V in delta cicuit, on secondary transformers windings voltages are too about 400 V in star circuit, but should be 230 V, because it's star circuit so voltages on secondary transformers windings is phase voltages (phase to neutral point voltage). On primary transformers windings it looks that is OK, because voltmeters connected to primary transformers windings show 400 V what is properly because primary transformers windings is connected in delta circuit so voltages on primary transformers windings are phase to phase voltages. Off course primary transformers windings and secondary transformers windings I used in proportions 1:1 for easy description for this problem. I hope that this problem will be repaired. Maybe I do some wrong, but I don't know so I wait on reply.

Mark.

pfalstad, 2020-04-15:
Hi, could you post a sample circuit?  Use File->Export As Link...

doomfan, 2020-04-15:
Paste all this what is below. I hope that I do it properly. Maybe this short link work ok too like http://tinyurl.com/so4pm9e      ?

http://falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+2.1831051418620846+50+5+43%0AT+256+144+368+176+0+4+1+0.06921878006633843+1.979077703487693e-16+0.999%0AT+256+208+368+240+0+4+1+0.46904117095272424+-1.2378627257623223e-16+0.999%0AT+256+272+368+304+0+4+1+-0.5382599510190551+-1.783053718470259e-17+0.999%0AR+256+144+208+144+0+1+50+230+0+0+0.5%0AR+256+208+208+208+0+1+50+230+0+2.0943951023931953+0.5%0AR+256+272+208+272+0+1+50+230+0+4.1887902047863905+0.5%0Aw+256+176+256+208+0%0Aw+256+240+256+272+0%0Aw+256+304+256+288+0%0Aw+256+288+272+288+0%0Aw+256+144+256+160+0%0Aw+256+160+272+160+0%0Aw+272+160+272+288+0%0Aw+368+304+384+304+0%0Aw+368+240+384+240+0%0Aw+368+176+384+176+0%0Aw+384+176+384+240+0%0Aw+384+240+384+304+0%0Ap+256+144+256+176+1+0%0Ap+256+208+256+240+1+0%0Ap+256+272+256+304+1+0%0Ap+368+144+368+176+1+0%0Ap+368+208+368+240+1+0%0Ap+368+272+368+304+1+0%0A

pfalstad, 2020-04-16:
Well, I don't know much about three-phase power.  But it seems that the voltage is the same on both sides of all the transforms, and that seems correct to me, since they're in 1:1 ratio.  So I'm not seeing the problem.  Is there a problem with the transformers, or somewhere else?

doomfan, 2020-04-16:
1:1 ratio should give same voltages on primary transformers windings and secondary transformers windings if primary and too secondary transformers windings connected in the same circuit - star circuit or delta circuit. Of course I don't mean here connecting physically primary and secondary transformers windings ! In sense, each of them separately should be connected in delta circuit or star circuit. Then 1:1 ratio should give the same voltages on both sides of transformers in three-phase system. But if you say that you don't know much about three-phase system I think that further discussion does not make sense. I can only say you that if it mean about three phase system it looks like it, read below if you interested.

delta-delta circuit, if 1:1 ratio voltages on both sides of transformers is the same (all voltages are phase to phase voltages)

star-star circuit, same like delta-delta star, if 1:1 ratio voltages on both sides of transformers are the 
same but all these voltages will be lower about 1,73 times (all voltages are phase-neutral point voltages)

delta-star circuit, if 1:1 ratio voltages on the side of secondary transformers windings should be lower about 1,73 times, because on the side of primary transformers windings voltages are phase to phase voltages and on the side of secondary transformers windings voltages are phase-neutral point voltages

star delta circuit, if 1:1 ratio voltages on the side of secondary transformers windings should be higher about 1,73 time, because on the side of primary transformers windings voltages are phase-neutral point voltages and on the side of secondary transformers windings voltages are phase to phase voltages

Generally on three-phase system - phase to phase voltage is about 1,73 times higher than phase-neutral point voltage. Off course I talk about symmetric three-phase system where all phase to phase voltages are the same (On Europe, about 400 V) and all phase-neutral point voltages are the same (On Europe, about 230 V). I guess that you are from The USA or Canada so european electric system is for you a bit strange. I am from Poland.

pfalstad, 2020-04-16:
Here it says that in a delta-star circuit, the secondary should have voltage that is 1.73 times the primary voltage: https://www.electronics-tutorials.ws/transformer/three-phase-transformer.html

That is what we're seeing in this circuit..  The input voltage sources are all 230 V, and the output is about 398 V, which is 230 V * 1.73.

doomfan, 2020-04-16:
I reccomend you write about it from link below. This is not for sure simply. Rather think that I'm right, but you can rate it yourself. Use Google Translate, because its this PDF is in polish language. These are (I suppose) some materials from any polish college on this subject.

http://www.witczak.imsi.pl/images/wyklady/maszyny_elektryczne_1/7_transformatory_trojfazowe.pdf

mopfeil, 2020-04-16:
I believe your pdf is not saying anything different than the document Paul
referenced. The Line Voltages get factors of Root(3) up or down, for
delta-star and star-delta,  the phase voltages remain the same. I do not
see a problem.

doomfan, 2020-04-16:
@mopfeil , Did you read this PDF and this site from links given by me in my previous comment ? I did read these materials and it looks for me that however I am right. In these materials is exactly "about 180 degrees" otherwise than in material wchich was given by Paul Falstad in his last comment.

But who really say right about it ? Think logically. If you connected primary transformers windings as delta circuit what will you get on secondary transformers windings ? According to me the same like on primary transformers windings, I mean here that:
1st transformer winding - voltage between phase 1 and phase 2
2nd transformer winding - voltage between phase 2 and phase 3
3rd transformer winding - voltage between phase 3 and phase 1
However, if you will connect secondary transformers windings as star circuit what will you do at this moment according you ? Think. According to me you will connect then voltage electric potentials of phase 1, phase 2 and phase 3 in one point called "neutral point". And what will you get if you add voltage electric potentials of phase 1, phase 2 and phase 3 according to you ? Off course remember now that all three voltage electric potentials of phase 1, phase 2 and phase 3 are shifted about 120 degrees, it mean that phase 2 is shifted about 120 degrees than phase 1, phase 3 is shifted about 120 degrees than phase 2 and phase 1 is shifted about 120 degrees than phase 3. Off course it mean too that phase 3 is shifted about 240 degrees than phase 1, phase 1 is shifted about 240 degrees than phase 2 and phase 2 is shifted about 240 degrees than phase 3. So I come back to my question for you - what will you get if you add voltage electric potentials of phase 1, phase 2 and phase 3 according to you ? According to me you will get 0 volts, because sum of voltage electric potentials of phase 1, phase 2 and phase 3 is 0 volts. Now come back to a few text lines above in this post, there where I wrote about delta circuit on primary transformers windings and star circuit on secondary transformers windings. Maybe you will understand what about mean here.

Mark.

mopfeil, 2020-04-16:
I will try to dig deeper tomorrow. Down in work today. 

————————————-
Markus Pfeil
+49 1512 7575725
m.o.pfeil@gmail.com

pfalstad, 2020-04-16:
From equation 7.6c in your www.witczak.imsi.pl pdf link, it seems the voltage of the secondary for a Dy configuration should be sqrt(3) times larger than the primary...  That assumes U2N is the secondary voltage.

The bezel.com link says the same thing in section 5.5, assuming U2 is the secondary voltage.

doomfan, 2020-04-16:
If you mean equation 7.6c click on link to see screen below. See especially these things marked by me red color.

[image]

pfalstad, 2020-04-16:
Ok, so it's a Dy right?  So we should use 7.6c?

What's theta N, it looks like that's primary voltage divided by secondary voltage?

doomfan, 2020-04-16:
"N" mean number of transformer turns. N1 is primary transformer turns, N2 is secondary transformer turns. I will take any old three-phase transformer (if I will get it somewhere, may be different) and I will check it myself, because I see, that we don't understand.

pfalstad, 2020-04-16:
ok but I was asking about "theta n", not "n".

mopfeil, 2020-04-17:
I have tried to follow Google translate but that is hard work.
However, looking at the diagram, I believe (may be wrong) that you had your
primary side wired up in reverse to the paper. I think you are going from 1
to 3 on the upper terminal on the primary side, instead of from 1 to 2 as
in the paper. This would explain the 180 degree phase shift. I have wired
it as I think it is in the paper, and the phases are, as expected flipped
by 180 degrees and should therefore be in alignment with your paper.
I also think that your papers is in alignment with Pauls resource.

http://falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+2.1831051418620846+50+5+43%0AT+256+112+368+144+0+4+1+0.4106872421720142+5.384533394887117e-17+0.999%0AT+256+208+368+240+0+4+1+0.5233826375520699+1.9726257955188826e-17+0.999%0AT+256+304+368+336+0+4+1+-0.3093431187417566+-1.9513376311783763e-17+0.999%0AR+256+112+208+112+0+1+50+230+0+0+0.5%0AR+256+208+208+208+0+1+50+230+0+2.0943951023931953+0.5%0AR+256+304+208+304+0+1+50+230+0+4.1887902047863905+0.5%0Aw+368+144+384+144+0%0Ap+256+112+256+144+1+0%0Ap+256+208+256+240+1+0%0Ap+256+304+256+336+1+0%0Ap+368+112+368+144+1+0%0Ap+368+208+368+240+1+0%0Ap+368+304+368+336+1+0%0A207+256+112+256+96+4+1-%5Cp%0A207+256+240+256+256+4+1-%5Cp%0A207+256+336+256+352+4+2-%5Cp%0A207+256+208+256+192+4+2-%5Cp%0A207+256+304+256+288+4+3-%5Cp%0A207+256+144+256+160+4+3-%5Cp%0Aw+384+144+384+240+0%0Aw+384+240+368+240+0%0Aw+384+240+384+336+0%0Aw+384+336+368+336+0%0Ao+3+128+0+4099+640+1.6+0+10+3+3+4+0+4+3+5+0+5+3+10+0+11+0+12+0+10+0%0A

closed 2021-12-13 by pfalstad