sharpie7/circuitjs1 #436: Spanish Traslate

xmsilva, 2020-04-18:
The following file is the Spanish translation, could you add it in the circuits program?
**[locale_es.txt](https://github.com/sharpie7/circuitjs1/files/4496947/locale_es.txt)**
I have tested it with the desktop version and it works correctly.
I use Circuits with my electronics and electricity students and for them it would be better if the program is in Spanish.
Thanks for this great code ...

JFalstad, 2020-04-18:
This is great, thank you!  I will put this in on Monday.  My web site is in the process of being migrated.

pfalstad, 2020-04-20:
Ok it's in place, you can view it here:

http://www.falstad.com/circuit/circuitjs.html?lang=es

later I will add it to the options dialog.

Thanks much!!  How should I give you credit in the about box?

xmsilva, 2020-04-21:
My students will be very happy with this possibility of accessing the program in Spanish.
In the about box :
J. Miguel Silva for the Spanish Translation.

Thanks so much!!

closed 2020-05-16 by sharpie7

blitobyte, 2025-01-16:
Hi, I've just come across your program and I like it very much. Actually, up to now, I used SimulIDE (which is really good) but some important components like JFETs are missing and I needed them for a course I'm doing, so I downloaded the local version of falstad app and it's the one I use now.
I've got the Spanish version, but, in my opinion, there are a few issues, at least from a Spanish point of view, and I'd like to correct them, if possible. I downloaded the 'local_es_V1.0.txt' and 'local_es_Modificada_AR.txt', and both have similar issues.  So, if I make changes on any of the translation files how can I incorporate them to the local falstad app?
How can I change the locale on the local app, I mean in case I wanted to use the English version?

Thanks a lot.

Pablo

pfalstad, 2025-01-16:
You can use Options->Other Options...  and then change the language.

I don't know what files you're referring to.  local_es_Modificada_AR.txt ?  I don't know which file you mean.

The most current versions of the locale files are here:
https://github.com/pfalstad/circuitjs1/tree/master/src/com/lushprojects/circuitjs1/public

If you send me the files I can put them in place so you can test them.  That is probably the simplest solution.  Otherwise, you have to do a build, which is pretty complicated.

pfalstad, 2025-01-16:
Hi @blitobyte ,

You can use Options->Other Options... and then change the language.

I don't know what files you're referring to. local_es_Modificada_AR.txt ? I don't know which file you mean.

The most current versions of the locale files are here:
https://github.com/pfalstad/circuitjs1/tree/master/src/com/lushprojects/circuitjs1/public

If you send me the files I can put them in place so you can test them. That is probably the simplest solution. Otherwise, you have to do a build, which is pretty complicated.

blitobyte, 2025-01-17:
Hi,

**_You can use Options->Other Options... and then change the language._** Silly me!! I hadn't seen this opion.

 **_local_es_Modificada_AR.txt_**.  I came across this file in a different thread about Spanish translation

Ok. I now have the latest locale_es. Unfortunately it still contains 'strange issues'. I'll rewrite it and send it to you. It could be that the previous locale_es was translated by an Argentinian (which is Ok, of course!), but weird in Spain's Spanish.

blitobyte, 2025-01-17:
Hi,
I'm sending you a first draft of the translation. I'm not an expert neither in electronics  nor in language translation. What I've done is 'polish' the text from a native speaker's point of view getting rid of 'false friends' with other languages (portuguese for instance). I'm sure that from a translation expert's point of view there might be inconsistencies in capitalization, punctuation and other linguistics aspects.
In the list you sent me, sometimes some context is missing what makes the translation a bit difficult. Once the translation is incorporated I think I might tune it up a bit while using the program.
The new locale will be operative in both the web version and the local version, won't it?

[locale_es_ES.txt](https://github.com/user-attachments/files/18452579/locale_es_ES.txt)

pfalstad, 2025-01-17:
Ok, you can try it here:

https://www.falstad.com/circuit/circuitjs.html?lang=es_ES

I think "archivo" should be capitalized.

So you think the old file is fine for Argentina, but you want to use this one for Spain?

Yes, it can be used in both the web version and offline version.

pfalstad, 2025-01-17:
Ok @blitobyte , you can try it here:

https://www.falstad.com/circuit/circuitjs.html?lang=es_ES

I think "archivo" should be capitalized.

So you think the old file is fine for Argentina, but you want to use this one for Spain?

Yes, it can be used in both the web version and offline version.

blitobyte, 2025-01-17:
Hi @pfalstad ,

Ok, I'll revise the capitalization, it was a first try and I tink it needs some more goes at it. To be honest I didn't expect you to answer so quickly. ;-)

Capitalization in Spanish is very different from English, so words like 'Archivo', 'Todos', 'Salvar' shouldn't be in capitals unless there's a good reason for it. If they begin a sentence, ok. But don't worry, I'll have a deeper revision.

Actually I overread the Argentinian translation very quickly. There were a few things like the use of  'Rótulo' which for me are strange. A 'Rótulo' rather refers to banners, city signals. I'd say it's a big, big label.
Another word which is not definitely correct is 'chave' for switch. The appropriate word is 'interruptor', 'conmutador' or 'pulsador' depending on the type of switch. 'Chave', in fact doesn't exist in Spanish. It's the pronounciation of the word 'llave' (key) in Argentinian Spanish. They pronounce words like 'llave (key)', 'llover (rain)', llevar (take)' as 'shave/chave', 'shover/chover', ' shevar/chevar'. Definitely not acceptable.!
If you want I can have a deeper reading at the Argentinian translation just in case there's something rather weird, always considering that localisms are perfectly ok, and for Argentians I'm sure my Spanish is full of localisms. ;-)

One more thing: there's no way to edit and modify the text on the fly and see the results without having to bother you sending changes constantly?

blitobyte, 2025-01-17:
Hi again @pfalstad,

I've just had a look at the web version, and now I understand what you meant about 'Archivo'. In that case you were right, it must be in capitals. 
That's why I asked you about there being a quickest way to change and check. As I mentioned, the list of words don't indicate where the translation is used.

By the way there are options which don't appear in the locale, like 'Add Realy Coil', Add Photoresistor'... etc. They appear in English in the Spanish version.

pfalstad, 2025-01-17:
Hi @blitobyte ,

I'm just wondering whether I should get rid of the old Spanish file, or keep using it for Argentina / Latin America.

Here is a list of strings missing from the Spanish strings file:

https://github.com/pfalstad/circuitjs1/blob/master/lang/missing_es.txt

If you want to quickly change and check, I think you have to do a build.  You would need to clone a copy of the respository first. 
 Then you can go here for info:

https://github.com/pfalstad/circuitjs1/blob/master/README.md

Skip down to "Development using Gradle"..  I think that's the simplest.

If that is too complicated, you can just send the string files to me.  I don't mind.

blitobyte, 2025-01-18:
Hi,

About the old Spanish locale I wouldn't keep using it because it contains quite a few issues as I mentioned. It's up to you to leave it for Argentina/Latin America.

About the list of missing strings, what shoul I do? Add it to the main locale file?

About rebuilding the simulator I have a couple of questions:

1. The local version is made in Java?
2. If so, it shouldn't be a problem because I've worked with Java for more than 20 years (with Netbeans). You suggest Eclipse (which I used long ago, Eclipse 3 I think), also suggest VSCode which I've being using in the last 10 years mainly with Python and C/C++, and recommend Gradle which I've never used. I haven't used any of the three with Java, so should I go for VSCode which is the closest to me? or go for Gradle being it your reccomendation? If so, I wouldn't mind if it's the simplest and quickest way to go.

pfalstad, 2025-01-18:
@blitobyte ,

Actually I just realized it's much easier to test the strings files!  Just put them in circuitjs1/resources/app/war/circuitjs1/locale_es.txt in the offline version.  I assume you're using windows?  If you're using a mac, put them in CircuitJS1.app/Contents/Resources/app/war/circuitjs1/locale_es.txt

For the list of missing strings, add them to the file.  For example, for "Add Crystal", just add a line to your locale_es.txt file like this:

"Add Crystal"="Add Crystal in Spanish"

blitobyte, 2025-01-18:
Hi @pfalstad ,

I use linux: Debian 11 and Xubuntu 24-04 (right now this). I have the local version at: /home/pablo/opt/falstad/.

Great!! I just put the new revised locale_es.txt at _war/circuitjs1_ and it worked!! much better now because I've revised the translation (a very complete revision) and there were a lot of errors, most of them mine because I had the brilliant idea of doing  lots of 'Find and replace' actions  without considering capitals and things of the sort. Now I've gone line by line and I think it's much better. 

Despite all, I'll keep revising the file as I use the simulator. I'm sure there'll still be some issues with the translation, but what worries me most is my 'ignorance' in electronics (I'm a hobbist) and I might confuse electronic concepts and terms, as well as habitual 'English' phrases to refer to certain concepts. Anyway, I'll make it.

I'm sending you a copy of locale_es.txt to replace the previous one although it doesn't include the missing terms. As soon as I can I'll send you the whol

[locale_es.txt](https://github.com/user-attachments/files/18464102/locale_es.txt)

e file.

blitobyte, 2025-01-18:
Hi @pfalstad,

Just to tell you that I have already a quite operative Spanish version. There are a few thins, though, I'd like to comment with you.

First I converted every line from the extended file to something like this:

...
"Add External Voltage (JavaScript)"="Add External Voltage (JavaScript)!!"
"Antenna (amplified)"="Antenna (amplified)!!"
"Base Inductance (H)"="Base Inductance (H)!!"
...

then  I merged the extended message file with the original locale_es.txt.  And the run the simulator. Every time I came across a label ending in !! I could understand the context and proceed accordingly.
After, theoretically, having no !!-labels I discovered there were 204 unused labels from the extended file. Does that make sense to you? I have probably missed some occurrences in the simulator, but so many?

Apart from that, there are some literals in the simulator that don't exist in the extended file. I suppose
 they are hard-coded in the program, aren't they?

Missing literals:

Add DPDT Switch
Add Relay Coil
Add Relay Contact
Add Cross Switch
Add 3-Phase Motor
Add Motor Protection Switch
Toolbar
Subcircuits...
Mouse wheel sensitivity
Minimum Target Frame Rate
carrier (in Scope)
Subcircuit Manager
Delete (in Subcircuits)
Done (in Subcircuits)
Mode:
Confirm window messages

I'm sending you the unused literals in a file.

[unused_extended_literals.txt](https://github.com/user-attachments/files/18466107/unused_extended_literals.txt)

pfalstad, 2025-01-18:
Yes, that list of missing literals was a bit old.  I have added a lot of new strings since then.  The script that generates the list of missing strings is broken at the moment.  I will fix it later.  For now, you can just translate any English strings you see.

"Add Crystal" is in the menus under Draw->Passive Components, but you can just hit the slash key "/" and search for it there.  Most of the other "Add ..." strings are there too.  Also "crystal" is shown if you create a crystal and mouse over it.

"Adjust timestep..." gets printed if you add an Audio Input and then select a .wav file.

"B-C Leakage ..." gets shown if you create a transistor, right click, edit, and "Create New Model".

"Chaos 1" is under Circuits -> Op Amps -> Chaotic Circuits.  A bunch of other strings are in the menus too.

"JK=..." is in Circuits->Sequential Logic->Flip Flops ->JK Flip Flop.

"VCVS" is shown in the lower right if you create a VCVS and mouse over it.

"melted" is shown if you mouse over the fuse in this circuit: https://tinyurl.com/2ce5lxv9

Most of the rest of these are still there.  Keep looking!

pfalstad, 2025-01-18:
@blitobyte 

Yes, that list of missing literals was a bit old.  I have added a lot of new strings since then.  The script that generates the list of missing strings is broken at the moment.  I will fix it later.  For now, you can just translate any English strings you see.

"Add Crystal" is in the menus under Draw->Passive Components, but you can just hit the slash key "/" and search for it there.  Most of the other "Add ..." strings are there too.  Also "crystal" is shown if you create a crystal and mouse over it.

"Adjust timestep..." gets printed if you add an Audio Input and then select a .wav file.

"B-C Leakage ..." gets shown if you create a transistor, right click, edit, and "Create New Model".

"Chaos 1" is under Circuits -> Op Amps -> Chaotic Circuits.  A bunch of other strings are in the menus too.

"JK=..." is in Circuits->Sequential Logic->Flip Flops ->JK Flip Flop.

"VCVS" is shown in the lower right if you create a VCVS and mouse over it.

"melted" is shown if you mouse over the fuse in this circuit: https://tinyurl.com/2ce5lxv9

Most of the rest of these are still there.  Keep looking!

Could you email me directly at paul at falstad dot com?  I keep forgetting to @ you..  and also the other people involved in this issue must be very annoyed.