sharpie7/circuitjs1 #437: Too high threshold voltage

MarinoHappiness, 2020-04-20:
Hello,

The default value of MOSFET threshold voltage is 1,5 V. However, when MOSFET is connected to higher voltages (100V), more voltage is needed to fully close the circuit. The ordinary 5V logic doesn't work any more.

[http://tinyurl.com/y83ebeud](http://tinyurl.com/y83ebeud)

MarinoHappiness, 2020-04-22:
The MOSFET didn't close up until the gate voltage reached 100V. That was strange because MOSFETs don't need such a high voltage to close the circuit. 

You don't need to change anything; keep adding new components. Is it possible in future to implement mathematical elements for building complex models (something we see in Matlab)?

Felthry, 2020-07-21:
I think your problem, @MarinoHappiness, is that you're trying to use an N-channel FET as a high-side switch with no special drive circuitry. Remember that the voltage that matters is Vgs.

pfalstad, 2020-12-11:
Closing since we don't need to change anything.

closed 2020-12-11 by pfalstad